buxtronix/am43 #1: Wifi Error? (probably user error but no idea where)

j5jonny5, 2020-06-07:
Hi this is exactly what I was looking for cant thank you enough for the time taken to make it, when verifying in arduino for upload I get the error below. Have specified the settings in the config.h. Sorry if this is painfully obvious to an experienced user, just starting out on my HA and esp32 adventure.

am43-client:222: error: variable or field 'WiFiEvent' declared void

 void WiFiEvent(WiFiEvent_t event) {

                ^

am43-client:222: error: 'system_event_id_t' was not declared in this scope

C:\Users\Admin\Downloads\am43-master\AM43Client\AM43Client.ino: In member function 'boolean AM43Client::connectToServer()':

AM43Client:239: error: 'serviceUUID' was not declared in this scope

   BLERemoteService* pRemoteService = m_Client->getService(serviceUUID);

                                                           ^

AM43Client:248: error: 'charUUID' was not declared in this scope

   m_Char = pRemoteService->getCharacteristic(charUUID);

                                              ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: E:\Games\Arduino\libraries\WiFi
exit status 1
variable or field 'WiFiEvent' declared void

buxtronix, 2020-06-08:
I think the issue here is that the Arduino IDE is compiling the files in the wrong order. I'll have a play with this, it seems that it just concatenates the .ino files and compils, but your ordering may be different from mine.
I'll work on a PR to get this consistent.

buxtronix, 2020-06-09:
Please re-pull the repository and try again. I believe I have things setup per the proper Arduino IDE behaviour.

j5jonny5, 2020-06-09:
Complied and loaded onto the esp perfectly thank you. 
Reading the MQTT states easily enough through the mosquito broker and dev tools. Struggling to publish the commands to the device or create the HA integration, time to hit youtube and work out how to manually configure the device.

j5jonny5, 2020-06-09:
100% working. Thank you for the support. excellent job.

closed 2020-06-09 by j5jonny5