buxtronix/am43 #2: AM43 controller keeps stopping

Kite-Smurf, 2020-06-15:
AM43 controller keeps stopping when multiple clients are in range. (three at this time) log files available here: [Download link](https://we.tl/t-MUD0vn2p6o)

buxtronix, 2020-06-15:
I'm seeing an exception and backtrace in the logs.

[17:51:42.296] - [COM4] - [>>] - abort() was called at PC 0x401b7267 on core 1
[17:51:42.303] - [COM4] - [>>] - 
[17:51:42.303] - [COM4] - [>>] - Backtrace: 0x40091f40:0x3ffcec00 0x40092171:0x3ffcec20 0x401b7267:0x3ffcec40 0x401b72ae:0x3ffcec60 0x401
b61ff:0x3ffcec80 0x401b62f2:0x3ffceca0 0x401b62a9:0x3ffcecc0 0x400d5a2b:0x3ffcece0 0x400d44ce:0x3ffced20 0x400d32d6:0x3ffced40 0x400dc829
:0x3ffcedd0 0x4008e689:0x3ffcedf0

Could you help me by installing the ESP exception decoder in Arduino and running the backtrace through it? (you might have to recompile/upload once the decoder is installed).

https://github.com/me-no-dev/EspExceptionDecoder

Kite-Smurf, 2020-06-16:
Hi Thanks for help!

This is the translation of the backtrace:

Decoding stack results
0x400943b0: invoke_abort at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c
line 155
0x400945e1: abort at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c
line 170
0x401bae77: __cxxabiv1::__terminate(void (*)()) at
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc
line 47
0x401baebe: std::terminate() at
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc
line 57
0x401b9e0f: __cxxabiv1::__cxa_throw(void*, std::type_info*, void
(*)(void*)) at
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc
line 87
0x401b9f02: operator new(unsigned int) at
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc
line 54
0x401b9eb9: operator new[](unsigned int) at
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc
line 32
0x400d5e5f: WiFiUDP::parsePacket() at
C:\Users\otto.elter\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiUdp.cpp
line 210
0x400d4802: ArduinoOTAClass::handle() at
C:\Users\otto.elter\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src\ArduinoOTA.cpp
line 367
0x400d358a: loop() at
C:\Users\otto.elter\Desktop\am43-client/am43-client.ino line 360
0x400dd5d9: loopTask(void*) at
C:\Users\otto.elter\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp
line 19
0x40090af9: vPortTaskWrapper at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c
line 143

I hope this helps.

Kite-Smurf, 2020-06-16:
What I also noticed that the controller crashes at:
- Attempting to connect to: 02:cc:5a:a4:e2:76  (any of the addresses)

and sometimes it is able to reboot automatically:

Attempting to connect to: 02:87:4f:1d:a4:97 ...abort() was called at PC
0x40087969 on core 0

this happens also:
 - Attempting to connect to: 02:bd:8e:b0:58:b1 ...[02:87:4f:1d:a4:97] Got
position: 100
 - [02:87:4f:1d:a4:97] Disconnected
then it crashes.

buxtronix, 2020-06-17:
I think this is caused by general instability in the way Arduino has configured the bluetooth stack.

After a lot of debugging, etc, I have managed to make significant improvements in the BLE
stack and two different BLE codebases i'm working on are running nearly rock-solid now.

The main fix is just to not have Arduino start the "Bluetooth classic" controller, but only run in BLE mode. Unfortunately most of this change is within the Arduino HAL, so needs to be patched yourself.

Please grab the latest version of this repo, and also see the updated manual patches in the README.

ccmtozzi, 2020-06-18:
Hi, I'm having a similar issue! I did installed the last version and patch the BLE as suggested in the README. Not much used to Arduino but wanted to give it a try! It work for about a minute and stops! I do have 6 devices that are nearby, I was using another python script but liked the idea to move away from my RPI W to ESP32!! 

[log.txt](https://github.com/buxtronix/am43/files/4796000/log.txt)

buxtronix, 2020-06-18:
```
Attempting to connect to: 02:d1:ce:cd:e7:68 ...ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

```
This suggests a power supply issue.

Also, the Arduino distribution has a limit of 3 devices it can connect to:

https://github.com/espressif/arduino-esp32/issues/3283

Try testing with one of the motors turned off to see if that is a cause.

ccmtozzi, 2020-06-18:
Thanks!! I will try to turn off 3 of the 6 and see if it works! I did test connected to my computer and in another USB power supply and same happened, probably it freezes when connecting to the 4th device. Since there is this 3 device limitation I wonder if would be possible to add to the code the devices it connects to instead of search for it, with that I could have 2 ESP32 that would connect to 3 devices each, maybe something around the code where you identify if the device has the service AND its one of the MAC in the list!! I wish my c++ was decent enough to try! :)

buxtronix, 2020-06-18:
Yes, it seems reasonable to add a whitelist of mac addresses or names to connect to. Stay tuned!

buxtronix, 2020-06-18:
I've restructured the repo as a library and also added whitelisting of addresses. Please check the readme for new insructions and give it a go.

Kite-Smurf, 2020-06-20:
Thanks for updating the sketch.
The white list seems to be a very good option, however I'm unable to get it working with 3 motors. Leaving the white list empty would make the ESP32 connect to all three motors, but the controllers stops working soon. I do not see any backtrace issues anymore. 
The controller stops usually at this line:
- Attempting to connect to: 02:cc:5a:a4:e2:76 ...[02:cc:5a:a4:e2:76] Disconnected
and also would not restart automatically.

Using the white list makes the ESP32 to connect properly but only to two motors even if I whitelist 3 MAC addresses.
#define DEVICE_ALLOWLIST "02:cc:5a:a4:e2:76,02:bd:8e:b0:58:b1,02:87:4f:1d:a4:97"
I'm not sure why, but the MAC in the middle always get disregarded. The log says "not included in the white list"
The controller does not produce any disconnecting issues or crashes having only two MAC's in the white list.

Kite-Smurf, 2020-06-20:
A few other questions regarding the Home Assistant side.

1)  I found this remark in the sketch: * - am43/<device>rssi       - The current RSSI reported by the device.
Is this still an option to use? I cannot seem to get any traffic on this topic.

2) the Covers and motor states keep showing offline states. ESP32 restart is needed to update the availability states. Is this caused by the controller or my Home Assistant setup?

3)  If the AM43 controller goes offline, none of its related entities go in offline or unavailable state.

4) In case one of the motors go offline, its light and battery entities show online state. If the motor is powered on again, its state and its cover state would not update in Home Assistant as long as the ESP32 is not restarted.

5)  Would it be possible to setup the ESP32 sketch so that we can make use of the Auto Discovery function of Home Assistant?

buxtronix, 2020-06-21:
1. Allowlist should be fixed now, I had a bug in the list parsing code.

2. rssi is not yet implemented - the BLE library RSSI function seems unstable, I need to look into it further.

3. The offline states are due to a MQTT limitation of only one LWT topic per client - fixing these would require implementing a separate MQTT client per AM43 connection. May be feasible, I shall look into it.

4. Auto discovery looks interesting, it's something I can look into for sure.

buxtronix, 2020-06-21:
Also, the persistent offline states are due to message retain settings. Rather than fix that in particular, I'll consider updating to one MQTT connection per AM43.

buxtronix, 2020-06-21:
Please try the code in the 'mqtt-multiclient' branch. This uses a separate MQTT connection per AM43, allowing proper availability.

You can also copy the 'availability_state' from the cover entity over to the battery/light entities to get availability for them.

buxtronix, 2020-06-23:
I've had good stability with this so merged the MQTT changes into master.

Kite-Smurf, 2020-06-23:
I still cannot get it working with three motors, only with max two. Also,
with the two I'm having range issues. If the ESP32 is farther than a meter
from the motors, the controller stops regularly and needs restarting. I've
just ordered a new ESP32 to see if that is causing the issue. I'll keep you
updated.

buxtronix, 2020-06-23:
I only have 2 devices to test, so don't know why 3 is failing.

As for range, sounds like the antenna might not be connected properly. One of my devices didn't have the antenna selection jumper in the right place and also got only a metre.

Kite-Smurf, 2020-06-23:
I'll do the testing for you with three devices, no problem at all. As soon
as I get my new EPS's in, I'll do some more testing as the "not working
with three devices" problem might be related to the range issue.

Kite-Smurf, 2020-06-26:
Finally, my new EPS's arrived to do some more testing.
I setup one with external antenna connected to the ipex, jumper re-positioned accordingly. The connection and the stability got 100 times better then in case of using the on board antenna.
According to the documentation the range should be much better than this even with the on board antenna. I came across some post mentioning the option to change the TX power of the BLE. do you think it is possible?
https://community.openmqttgateway.com/t/esp32-ble-range/249/11

Please find the logs here:
Download link
https://we.tl/t-I4ZC6e816K

buxtronix, 2020-06-28:
@Kite-Smurf Can you confirm you've made the changes to BLEClient.cpp per the README? A few of the crashes look similar to what happens without the patch.

Kite-Smurf, 2020-06-28:
BLEClient.cpp is pacthed according to README just as esp32-hal-bt.c . Please check if this is correctly done.
Download link
https://we.tl/t-mUcYn3mhj1

Please let me know if you need more details.

buxtronix, 2020-06-29:
I think this is at least in part due to a memory leak. I found one cause to this and will upload the fix shortly. You can see the improvement in free memory availability here:

[image]

I'll update when I've uploaded the fix.

buxtronix, 2020-06-29:
The fix is submitted, this should hopefully prevent most/all of the freeze you are seeing.

adotagarwal, 2020-06-30:
I also experience freezing. I tried the latest code and it didn't help.  I was previously using an older build which seemed to work a bit more stable but I suspect the issue has to do with the number of devices.

In short, my set up is 3 motors added to the configs allow list. I also get either an abort() or the chip just freezes. It is highly reproducible in my environment when all 3 sensors are connected. I believe it was also happening with just 1 sensor, but perhaps less often.

For example, a stop might occur:

BLE Advertised Device found: Name: , Address: 02:9c:d3:6b:11:5e, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
Attempting to connect to: 02:9c:d3:6b:11:5e ...

Or a restart:

[Bedroom M-R 02:87:42:46:69:88] Unknown reply: 9AA9100000001100000000010000110000000022
abort() was called at PC 0x401b7b1b on core 1

Backtrace: 0x40091f40:0x3ffcee10 0x40092171:0x3ffcee30 0x401b7b1b:0x3ffcee50 0x401b7b62:0x3ffcee70 0x401b69fb:0x3ffcee90 0x401b6aee:0x3ffceeb0 0x401b6aa5:0x3ffceed0 0x400d5273:0x3ffceef0 0x400d3d16:0x3ffcef30 0x400d2ee0:0x3ffcef50 0x400dcf4d:0x3ffcefe0 0x4008e689:0x3ffcf000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Starting Arduino BLE Client application...

buxtronix, 2020-06-30:
@adotagarwal Could you pass the backtrace through the [ESP exception decoder](https://github.com/me-no-dev/EspExceptionDecoder) and let me know traceback?

adotagarwal, 2020-07-01:
Sure here is one exception: 

PC: 0x401d284d: AM43Client::onDisconnect(BLEClient*) at D:\OneDrive\Documents\Arduino\libraries\am43-master\src\AM43Client.cpp line 36
EXCVADDR: 0x00040056

Decoding stack results
0x401d284d: AM43Client::onDisconnect(BLEClient*) at D:\OneDrive\Documents\Arduino\libraries\am43-master\src\AM43Client.cpp line 36
0x400d8a8d: BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEClient.cpp line 186
0x400d95d1: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEDevice.cpp line 167
0x4014ac1a: btc_gattc_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x401438d2: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4008e689: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

I had forked your repo and I believe that version, without any tweaks to the BLE library, worked pretty well when placed close to the blinds.

This latest version, with the mods, seems to improve the range dramatically but often crashes

adotagarwal, 2020-07-01:
So based on what I read about this particular error (LoadProhibited) and the EXCVADDR each time being really close to 0, it seems like perhaps the pointer is NULL and were trying to access a member of a null structure (the offset being close to 0).  

Is it possible that some sort of locking is required?

References: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html

Kite-Smurf, 2020-07-02:
The last code is running with 3 motors connected,  already for a few days continuously, without freezing. This is good news. I'm seeing short disconnections and re-connections with the MQTT server which I did not see before. They are really short like a second or two. For the rest it looks like solid now. However I think it would be worth mentioning that an external antenna is required to get the range better than a meter unless the power can be regulated by the code.

[image]

buxtronix, 2020-07-02:
@adotagarwal Did you apply the changes to the BLE library per the readme? These crashes would suggest that it might not have been done.

adotagarwal, 2020-07-02:
@buxtronix yes, at line 183 and 238 in bleclient.cpp i inserted the two conditional breaks which check the connection id. also in the hal c file, i changed the mode to ble as suggested (change cfg variable, change the initialize to pass BLE.

adotagarwal, 2020-07-02:
After that last hang, I reset the chip and it almost immediately crashed:

```
Attempting to connect to: 02:37:8d:55:9a:4a ...ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Starting Arduino BLE Client application...
Wifi connecting to: austinsapt ... 
[WiFi-event] event: 0
[WiFi-event] event: 2
AllowList contains the following device(s):
 Mac: 02:9c:d3:6b:11:5e 
 Mac: 02:87:42:46:69:88 
 Mac: 02:37:8d:55:9a:4a 
[WiFi-event] event: 4
BLE Advertised Device found: Name: , Address: 78:ad:07:2b:84:c2, manufacturer data: e00000d8ca82956a, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: , Address: 17:90:73:17:e4:c0, manufacturer data: 4c00090603330a2a0014
BLE Advertised Device found: Name: , Address: 18:d6:d8:bb:65:2a, manufacturer data: 0600010920029135cec705616ecd8281250bdf9ea1669fd5da210ffeee
BLE Advertised Device found: Name: , Address: 40:cb:c0:d8:a1:d0, manufacturer data: 4c0010050d14eab84e, txPower: 12
BLE Advertised Device found: Name: FITNESS, Address: f0:d4:f7:8f:2e:ae, serviceUUID: d973f2e3-b19e-11e2-9e96-f0d4f78f2eae
BLE Advertised Device found: Name: , Address: 61:c7:5a:ec:89:b0, manufacturer data: 060001092002a3b212c5b137be43887ca9ff292108748c08abccd39576
BLE Advertised Device found: Name: , Address: 15:0e:5e:d0:17:21, manufacturer data: 4c00090603300a2a0015
BLE Advertised Device found: Name: , Address: 5f:10:6a:42:e6:d2, manufacturer data: 4c001005131c33007b, txPower: 12
BLE Advertised Device found: Name: , Address: 24:fc:e5:b1:7a:e5, manufacturer data: 7500420401806024fce5b17ae526fce5b17ae401000000000000
BLE Advertised Device found: Name: , Address: 40:cb:c0:d9:de:20, manufacturer data: 4c0010050214228291, txPower: 12
[WiFi-event] event: 7
WiFi connected
IP address: 10.42.0.104
Attempting MQTT connection...connected
BLE Advertised Device found: Name: Bedroom M-R, Address: 02:87:42:46:69:88, manufacturer data: 0000, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
Attempting to connect to: 02:87:42:46:69:88 ...[02:87:42:46:69:88] Connected
 - Failed to connect.
BLE Advertised Device found: Name: , Address: 24:fc:e5:b1:7a:e5, manufacturer data: 7500420401806024fce5b17ae526fce5b17ae401000000000000
BLE Advertised Device found: Name: , Address: 15:0e:5e:d0:17:21, manufacturer data: 4c00090603300a2a0015
BLE Advertised Device found: Name: , Address: 40:cb:c0:d9:de:20, manufacturer data: 4c0010050214228291, txPower: 12
BLE Advertised Device found: Name: FITNESS, Address: f0:d4:f7:8f:2e:ae, serviceUUID: d973f2e3-b19e-11e2-9e96-f0d4f78f2eae
BLE Advertised Device found: Name: , Address: 17:90:73:17:e4:c0, manufacturer data: 4c00090603330a2a0014
BLE Advertised Device found: Name: , Address: 40:cb:c0:d8:a1:d0, manufacturer data: 4c0010050d14eab84e, txPower: 12
BLE Advertised Device found: Name: , Address: 04:69:f8:99:91:af, manufacturer data: 4c000906030000000000
BLE Advertised Device found: Name: , Address: 78:ad:07:2b:84:c2, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: , Address: 5f:10:6a:42:e6:d2, manufacturer data: 4c001005131c33007b, txPower: 12
BLE Advertised Device found: Name: , Address: 18:d6:d8:bb:65:2a, manufacturer data: 0600010920029135cec705616ecd8281250bdf9ea1669fd5da210ffeee
BLE Advertised Device found: Name: Bean, Address: 80:ea:ca:03:07:01, manufacturer data: aaaa80eaca030701, serviceUUID: 00001802-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: , Address: 02:9c:d3:6b:11:5e, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
Attempting to connect to: 02:9c:d3:6b:11:5e ...[02:9c:d3:6b:11:5e] Connected
 - Connected to server
Connect took 1113ms
BLE Advertised Device found: Name: FITNESS, Address: f0:d4:f7:8f:2e:ae, serviceUUID: d973f2e3-b19e-11e2-9e96-f0d4f78f2eae
BLE Advertised Device found: Name: , Address: 04:69:f8:99:91:af, manufacturer data: 4c000906030000000000
BLE Advertised Device found: Name: , Address: 15:0e:5e:d0:17:21, manufacturer data: 4c00090603300a2a0015
BLE Advertised Device found: Name: , Address: 18:d6:d8:bb:65:2a, manufacturer data: 0600010920029135cec705616ecd8281250bdf9ea1669fd5da210ffeee
BLE Advertised Device found: Name: , Address: 17:90:73:17:e4:c0, manufacturer data: 4c00090603330a2a0014
BLE Advertised Device found: Name: , Address: 61:c7:5a:ec:89:b0, manufacturer data: 060001092002a3b212c5b137be43887ca9ff292108748c08abccd39576
BLE Advertised Device found: Name: , Address: 5f:10:6a:42:e6:d2, manufacturer data: 4c001005131c33007b, txPower: 12
BLE Advertised Device found: Name: , Address: 78:ad:07:2b:84:c2, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
[ 02:9c:d3:6b:11:5e] AM43 Send: 00FF00009A170222B815
BLE Advertised Device found: Name: , Address: 40:cb:c0:d8:a1:d0, manufacturer data: 4c0010050d14eab84e, txPower: 12
BLE Advertised Device found: Name: , Address: 24:fc:e5:b1:7a:e5, manufacturer data: 7500420401806024fce5b17ae526fce5b17ae401000000000000
BLE Advertised Device found: Name: , Address: 02:37:8d:55:9a:4a, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
Attempting to connect to: 02:37:8d:55:9a:4a ...[02:37:8d:55:9a:4a] Connected
[... 23 lines cut ...]
A14     : 0x3ffef894  A15     : 0x3ffde648  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
[... cut ...]
0x4014ac1a: btc_gattc_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x401438d2: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4008e689: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

100% i have made the edits specified in the readme

MrBigleg, 2020-07-02:
System is now completely unstable for me also. Constantly restarting in the logs

adotagarwal, 2020-07-02:
@MrBigleg which version were you previously on? Did you also update the example code or just the main library code?

I think perhaps theres an issue with the example, and not the library? @buxtronix are you using the example code in your application?

I noticed that in the callback, many of the serial.printfs are passing toString() and not toString().c_str()

adotagarwal, 2020-07-02:
All of the crashes / reboots seem to occur on disconnect, while it is currently scanning and connecting to devices.

If and when it finally connects to all of the devices it seems to operate fairly stable.

MrBigleg, 2020-07-02:
@adotagarwal I have been keeping up to date on each incremental version for the last week, the last version was the memory leak fix.

I had a separate issue where the blind was opening if at position 0 but I think it was my error and a mistake in Node-Red. I have always had disconnect (device unavailable) issues. But have only been using this for about a week

adotagarwal, 2020-07-02:
i think you should change the example code so that it correctly prints the two log lines: `Serial.printf("Ignoring advertising device %s, already present\r\n", advertisedDevice.toString().c_str());` and 
`Serial.printf("Ignoring device %s, not in allow list", advertisedDevice.toString().c_str());`

with this change, the restarts are much less frequent (i've noticed in other projects too that passing garbage to serial print will often corrupt memory in the esp) 

I still get the issue with disconnect

Kite-Smurf, 2020-07-02:
" I still get the issue with disconnect, it seems like it sometimes gets stuck and sometimes crashes: It always seems to happen during discovery and while connecting to one device and disconnecting from another "

I was experiencing this many times until I attached an external antenna. And it still happens if the controller is moved farther away from the motors.

adotagarwal, 2020-07-02:
> " I still get the issue with disconnect, it seems like it sometimes gets stuck and sometimes crashes: It always seems to happen during discovery and while connecting to one device and disconnecting [...]

Thanks - i don't think this has anything to do with the range at all. First, the devices are able to be controlled consistently from the range at which I am debugging with console -- once they finally connect and make it past a discovery cycle. Furthermore, the devices still lock up and freeze when placed within a foot of the windows.

buxtronix, 2020-07-02:
I have a feeling the problem is within the BLE library, not the AM43 code. In fact, after poking around with the code paths with disconnect events, I think I may have discovered the issue.

Within BLEClient.cpp the code calls the following to update a PeerDevice map within BLEDevice:

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.cpp#L101
BLEDevice::addPeerDevice(this, true, m_appId);

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.cpp#L191
BLEDevice::removePeerDevice(m_appId, true);

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.cpp#L237
BLEDevice::updatePeerDevice(this, true, m_gattc_if);

When you look into the implementation of those functions, it's clear that the latter call should have 'm_appId' as the last parameter, not m_gattc_if.

Can you change this and test?

Also, there are some fixes in the latest repo update here, so worth re-fetching.

adotagarwal, 2020-07-03:
Let me try and integrate the latest changes, I do see the printfs have been fixed there, and try the patches mentioned above. 

@Kite-Smurf i ordered some 32us that have an external antenna and will try with those as well.

MrBigleg, 2020-07-03:
Just a short term update from me. I live in Thailand so most likely a different timezone to many here, irrelevant but when I woke up this morning the latest version with the additional BLE edit had just been put up.

It is now the end of the day and I have had no noticeable disconnects or erratic behavior, all day in my HA logs.

I will monitor over the weekend and share my experience.

Have a good weekend...

adotagarwal, 2020-07-03:
A few updates:
@Kite-Smurf - the esp32u's came and as suspected did not improve the situation at all. I now have some fairly impressive antennas with a little esp attached at the end. 

@buxtronix :  without this change my system ran fairly well overnight but by this morning had disconnected a few blinds. I have some new suspicions:

1. sometimes, this code is putting the actual blind controller into a bad state, it will keep blasting out the battery. This can only be corrected by opening the blind engine app and connecting to them through there.  This morning, when a few wouldn't open even after restarting, I connected to the debug console and within just a second or two saw:

```
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 100
[02:9c:d3:6b:11:5e] Got battery: 100
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 96
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 100
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 100
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 100
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 96
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 100
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
[02:9c:d3:6b:11:5e] Got battery: 97
```

2. the change to the parameter in the connected branch of the switch didn't improve the issue with freezing / restarting during scan.

adotagarwal, 2020-07-03:
Also, given the program just stops executing sometimes, could it be that there are momentarily more than 3 devices attempting to be connected to, despite having three in the allow list? Perhaps a duplicate ( I do see it trying the same sensor a few times in the same scan, despite updating BLEScan.cpp)  I believe I read that the behavior is to stop executing.

Separately, I think there is an issue where the blinds get into a state where you need to open simply open the BlindEngine app to "reset" them.

buxtronix, 2020-07-03:
@MrBigleg I see the following in the log:

`MQTT [am43/restart]1: 0`

Something is sending a restart/reboot command over mqtt it seems.

buxtronix, 2020-07-03:
@adotagarwal The issue with freezing during connect whilst still responding to BLE callbacks, it seems that something in BLEClient::connect is never returning. Maybe due to more than 3 trying to be connected to? Many of the bugs I've encountered so far have been within the BLE library, this could be another.

I can look at applying a patch to count connections and not attempt > 3, this may improve things. I'll do that over the weekend.
I had the battery message spamming issue at times, seems mostly when the power supply is plugged in and the battery is full.

buxtronix, 2020-07-04:
I've just committed the change limiting connections to three.

MrBigleg, 2020-07-04:
So I woke up this moning an it appears the blind has been coonected almost without issue all night just one short 5min disconnect.

I'm also see strange things in the serial monitor...

BLE scan complete.
[Blind 02:ba:58:19:2c:d3] AM43 Send: 00FF00009AA2010138
[02ba58192cd3] Got battery: 100
[Blind 02:ba:58:19:2c:d3] AM43 Send: 00FF00009AA701013D
[02ba58192cd3] Got position: 0
[Blind 02:ba:58:19:2c:d3] Unknown reply: 9AA80032
[Blind 02:ba:58:19:2c:d3] Unknown reply: 9AA9100000001100000000010000110000000022
BLE Advertised Device found: Name: , Address: 64:41:ca:7b:73:77, manufacturer data: 06000109210a40f70837c6d04d494e492d5043
BLE Advertised Device found: Name: , Address: 7d:27:32:ed:59:b3, manufacturer data: 06000109210aec3271666c6043524149472d4445534b544f50
BLE Advertised Device found: Name: , Address: 71:03:1a:db:84:61, manufacturer data: e000015cca500b36, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: , Address: dc:a2:66:5c:32:22, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc
BLE Advertised Device found: Name: MJ_HT_V1, Address: 58:2d:34:32:d7:1f, serviceUUID: 0000180f-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:6a:5b:5c, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: , Address: 74:19:1c:9f:20:4f, manufacturer data: 4c0010050918b50925, txPower: 12
BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:6a:51:5a, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb
BLE scan complete.
[Blind 02:ba:58:19:2c:d3] AM43 Send: 00FF00009AAA010130
[02ba58192cd3] Got light: 50
[Blind 02:ba:58:19:2c:d3] AM43 Send: 00FF00009AA2010138
[02ba58192cd3] Got battery: 100
BLE Advertised Device found: Name: , Address: 64:41:ca:7b:73:77, manufacturer data: 06000109210a40f70837c6d04d494e492d5043
BLE Advertised Device found: Name: , Address: dc:a2:66:5c:32:22, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc
BLE Advertised Device found: Name: , Address: 7d:27:32:ed:59:b3, manufacturer data: 06000109210aec3271666c6043524149472d4445534b544f50
BLE Advertised Device found: Name: , Address: 71:03:1a:db:84:61, manufacturer data: e000015cca500b36, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:6a:51:5a, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: , Address: 74:19:1c:9f:20:4f, manufacturer data: 4c0010050918b50925, txPower: 12
BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:6a:5b:5c, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: MJ_HT_V1, Address: 58:2d:34:32:d7:1f, serviceUUID: 0000180f-0000-1000-8000-00805f9b34fb
BLE scan complete.
[Blind 02:ba:58:19:2c:d3] AM43 Send: 00FF00009AA701013D
[02ba58192cd3] Got position: 0
**[Blind 02:ba:58:19:2c:d3] Unknown reply: 9AA80032
[Blind 02:ba:58:19:2c:d3] Unknown reply: 9AA9100000001100000000010000110000000022**

[image]

MrBigleg, 2020-07-04:
> am43/restart

[image]

Very strange, this was an inject node that I set up when I was first testing it, it should trigger on it's own. I've now disabled it. Similar to my other 'issue'that I closed. Very odd that Node-Red was triggering these without input.

buxtronix, 2020-07-04:
```
[Blind 02:ba:58:19:2c:d3] Unknown reply: 9AA80032
[Blind 02:ba:58:19:2c:d3] Unknown reply: 9AA9100000001100000000010000110000000022
```

This is fine, when the code requests position from the device, it returns two other packets containing the config. These are ignore now but expected.

adotagarwal, 2020-07-04:
@buxtronix so I picked up that change and it doesn't seem related to the cls size, however, i did more digging and here's what I found. 

I turned on info level core logging, and noticed that it always gets stuck after this log line:

```
BLE Advertised Device found: Name: Bedroom Right, Address: 02:9c:d3:6b:11:5e, manufacturer data: 0000, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
Trying to connect to Name: Bedroom Right, Address: 02:9c:d3:6b:11:5e, manufacturer data: 0000, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb [1 devices used]
Attempting to connect to: 02:9c:d3:6b:11:5e ...[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 3, GATT role: client
```

When I look in the code, I see that in BLEClient, it is waiting for a semaphore to be available before it will write connected. 

To me, this implies that despite CLS being well under 3, the resource which the semaphore is protecting is still being exhausted.

Since I don't have as much knowledge of the BLE stack as you do, are you able to extrapolate anything from this?

The line of code is: BLEClient.cpp:102

adotagarwal, 2020-07-04:
Here is the trace log:

```
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: clients (0x3ffba454), owner: <N/A> for clientsAll
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: clients (0x3ffba454), owner: clientsAll
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: clients (0x3ffba454), owner: clientsAll
Attempting to connect to: 02:9c:d3:6b:11:5e [V][BLEDevice.cpp:60] createClient(): >> createClient
[V][BLEDevice.cpp:66] createClient(): << createClient
[V][BLEClient.cpp:96] connect(): >> connect(02:9c:d3:6b:11:5e)
[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 3, GATT role: client
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3fff9030), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3fff9030), owner: connect
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3fff9030), owner: connect for connect
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... Unknown
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RegEvt (0x3fff9030), owner: connect
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: RegEvt (0x3fff9030), owner: <N/A>
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3fff94a4), owner: <N/A> for connect
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3fff94a4), owner: connect
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3fff94a4), owner: connect for connect
```

This says clearly its dying while waiting for OpenEvt

adotagarwal, 2020-07-04:
I looked at the core library in github, and it seems like maybe this exact issue has been addressed in the library there?

https://github.com/espressif/arduino-esp32/commit/d1134fd45ec82193c9a754599b5b0f4ec54dd618

adotagarwal, 2020-07-04:
I see this issue which purports to fix the behavior. https://github.com/espressif/arduino-esp32/pull/2974/commits/85a34cbd10e4b376563563aa4880d03cbab29129

In this code, I see that it calls `esp_ble_gattc_app_unregister` with m_gattc_id. In the patch we applied most recently, in connect, it passes m_appid:

```
bool BLEClient::connect(BLEAddress address, esp_ble_addr_type_t type) {
	log_v(">> connect(%s)", address.toString().c_str());

// We need the connection handle that we get from registering the application.  We register the app
// and then block on its completion.  When the event has arrived, we will have the handle.
	m_appId = BLEDevice::m_appId++;
	BLEDevice::addPeerDevice(this, true, m_appId);
	m_semaphoreRegEvt.take("connect");

	// clearServices(); // we dont need to delete services since every client is unique?
	esp_err_t errRc = ::esp_ble_gattc_app_register(m_appId);
```

Would this call need to be changed similarly to the most recent patch?

adotagarwal, 2020-07-04:
So I think I've mitigated this lockup. The issue is that it gets stuck waiting for an open semaphore when one never arrives. This could be due to poor range (as @Kite-Smurf noticed) or interference (depending on what devices are near by, apparently dozens in my case - that's manhattan for you) or from what I've seen, the motors themselves needing to be reset.

What I've done is patched the FreeRTOS to add a wait that takes in a timeout and made it 5 seconds. The reason I couldn't just use take and then give or timedWait is because there seems to be no other way to get m_value, which is needed in our case.

With this patch applied, the chip **never** locks up at all - occasionally things seem to drop in / out but it never freezes or locks up.

I also think the change you last had us made is a positive one, it seems to resolve some of the disconnect exceptions which were occurring. 

Here is my wait function for FreeRTOS.cpp/h:

```

FreeRTOS.h:43

		uint32_t 	wait(uint32_t timeoutMs, uint32_t timeoutVal, std::string owner = "<Unknown>");

FreeRTOS.cpp:82

/**
 * @brief Wait for a semaphore to be released by trying to take it and
 * then releasing it again.
 * @param [in] owner A debug tag.
 * @return The value associated with the semaphore.
 */
uint32_t FreeRTOS::Semaphore::wait(uint32_t timeoutMs, uint32_t timeoutVal, std::string owner) {
	log_v(">> wait: Semaphore waiting: %s for %s", toString().c_str(), owner.c_str());
	bool rc = false;

	if (m_usePthreads) {
		assert(false);
	} else {
		rc = ::xSemaphoreTake(m_semaphore, timeoutMs / portTICK_PERIOD_MS) == pdTRUE;
	}

	if (!rc)
		return timeoutVal;

	if (m_usePthreads) {
		assert(false);
	} else {
		xSemaphoreGive(m_semaphore);
	}

	log_v("<< wait: Semaphore released: %s", toString().c_str());
	return m_value;
} // wait
```

Here's my modified BLEClient.cpp

```
/**
 * @brief Connect to the partner (BLE Server).
 * @param [in] address The address of the partner.
 * @return True on success.
 */
bool BLEClient::connect(BLEAddress address, esp_ble_addr_type_t type) {
	log_v(">> connect(%s)", address.toString().c_str());

// We need the connection handle that we get from registering the application.  We register the app
// and then block on its completion.  When the event has arrived, we will have the handle.
	m_appId = BLEDevice::m_appId++;
	BLEDevice::addPeerDevice(this, true, m_appId);
	m_semaphoreRegEvt.take("connect");

	// clearServices(); // we dont need to delete services since every client is unique?
	esp_err_t errRc = ::esp_ble_gattc_app_register(m_appId);
	if (errRc != ESP_OK) {
		log_e("esp_ble_gattc_app_register: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return false;
	}

	m_semaphoreRegEvt.wait("connect");

	m_peerAddress = address;

	// Perform the open connection request against the target BLE Server.
	m_semaphoreOpenEvt.take(5000, "connect");
	errRc = ::esp_ble_gattc_open(
		m_gattc_if,
		*getPeerAddress().getNative(), // address
		type,          // Note: This was added on 2018-04-03 when the latest ESP-IDF was detected to have changed the signature.
		1                              // direct connection <-- maybe needs to be changed in case of direct indirect connection???
	);
	if (errRc != ESP_OK) {
		log_e("esp_ble_gattc_open: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return false;
	}

	uint32_t rc = m_semaphoreOpenEvt.wait(5000,ESP_GATT_ERROR,"connect");
	log_v("<< connect(), rc=%d", rc==ESP_GATT_OK);
	return rc == ESP_GATT_OK;
} // connect
```

I'm going to test this out some more and then commit these modified files to my branch.

I still need to examine things abit because it seems

buxtronix, 2020-07-04:
Thanks for all the analysis, you've dug around quite a lot!

So obviously the CONNECT event is not coming in. I'm very skeptical that *no* event is coming in, it's probably just something the logic does not handle. e.g Perhaps it gets a CLOSE or DISCONNECT. Your timeout may just be working around this type of issue.

BLEClient.cpp:157 has a debug line, but unfortunately it doesnt seem to show all events (I can see 'Unknown' in your debug logs). Perhaps it needs to print the raw event number (with %d) for every event, then when the connect is attempted, there should be something logged. For example if a ESP_GATTC_CLOSE_EVT came, it does a no-op.

adotagarwal, 2020-07-04:
I actually added a log to the CLOSE EVT after seeing everything was commented out. It doesn't get hit. This is super frustrating.  I could certainly try and find the Unknown line and add more info.

As far as the scan stopping when I time out the semaphore for OpenEvt, I see that in your code once a device is found, the scan stops and then the next loop will attempt to connect to that device, before starting a new scan. Is this correct? I think the reason the scan stopped with my change is because connect now returns false, and so it never resets the lastScanTime. 

I've kind of modified your code to allow the scan to fully complete before attempting to connect to any of the devices. I wonder if the scan itself is making the ble connections unstable. 

When the device manages to connect to all three blinds, I've seen it remain stable for a very long time. However, when the devices drop they almost never come back on their own. A combination of connecting from the app, waiting some amount of time, or resetting the ESP is necessary.  As a side note, I have another ESP which only connects to one blind and it has been rock solid for several days now.

I wonder if its really necessary to scan for devices every so often, or rather only on the startup. In startup, wouldn't it be possible to try to connect several times (I've seen this is necessary over and over from different libraries) and then proceed with the loop for the devices which were found? If a device is lost, it could initiate a new scan.

I've also started hedging my bet by coding a Python alternative for raspberry pi based on some very rudimentary code available here on github. The pi seems to get very good range, as long as its not enclosed.

One thing I noticed with the Pi, again this is probably a deficiency of the OS, driver, or python library, is that when a scan is conducted devices tend to drop.

adotagarwal, 2020-07-05:
I mentioned this above but its worth saying again, I played around with the python clients for a while and then saw your comment on this ticket so I decided to try out the ESP again. It connected right away to all three devices (after completing the scan) and remained solid UNTIL the Arduino OTA crashed the ESP (see below)

I think that when these motors experience a disconnect like this, they need to be somehow disconnected/reset or the slate cleaned and that will allow them to then be discovered again later.

```
Decoding stack results
0x40091f40: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40092171: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x401bbf87: std::length_error::length_error(char const*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/cow-stdexcept.cc line 68
0x401badeb: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 82
0x401baede: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 50
0x401bae95: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 45
0x400d5f53: WiFiUDP::parsePacket() at D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiUdp.cpp line 205
0x400d465e: ArduinoOTAClass::handle() at D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src\ArduinoOTA.cpp line 360
0x400d36a8: loop() at D:\OneDrive\Documents\Arduino\AM43 Blinds\MQTTBlinds-Fauxmo/MQTTBlinds-Fauxmo.ino line 462
0x400e1591: cbuf::write(char const*, unsigned int) at D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp line 171
0x4008e689: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

buxtronix, 2020-07-05:
I wonder if this is actually a heap memory issue? That crash kind of looks like it. xPortGetFreeHeapSize() can get the free amount, typically there should be a few dozen kb free at any time.

I'd still be interested in seeing if any of the event callbacks get triggered at all. BLEClient.cpp:157 should be logging some event assuming esp_ble_gattc_open returns successfully (or maybe the event is in the gap handler...)

The other curious issue is that you say when in this state they can't be discovered. Is this even after an ESP restart? Usually they dont send advertisement packets while connected to something, perhaps they still have a connection somewhere..
(Maybe a neighbour is running the code also and 'hijacking' your controller!)

The last option is to have two ESPs, one on 2 blinds and the other on one, they are fairly cheap :)

Anyway, enjoy the festivities and stay safe!

adotagarwal, 2020-07-05:
I'll try out that change to 157 and also have it print out the heapsize periodically and post back tomorrow.

When i say cant be discovered, I mean that they are found but can consistently never be connected to. They either time out (now) or they fail right away with the standard cant connect.

buxtronix, 2020-07-05:
Also about scanning...I've found that the ESP cannot connect to a device whilst actively scanning. This doesn't seem to be a BLE library issue, I also get it in ESPHome. Hence why I stop scanning before connecting.

There are still a couple of strategies to pick from - (1) scan for a time, then attempt to connect to all that are found, or (2) Immediately stop and try to connect to a device as it appears in the scan, then resume the scan after connect. Ever since I patched BLEScan.cpp, both have been equally effective for me.

[44 more replies]

closed 2021-02-12 by buxtronix