buxtronix/am43 #4: Compiling issues

frizbeano, 2020-06-21:
Not a big deal since I did get it to work in the end.

Error Message
**MQTTBlinds:33:24: fatal error: AM43Client.h: No such file or directory
compilation terminated.
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\BlaBla\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.39.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
AM43Client.h: No such file or directory**

**Steps**

I copied over the **src** files into the **examples** folder and compile

Workaround
Changed 
#include <AM43Client.h>
to
#include  "AM43Client.h"

Which looks into the current **DIR** instead of the lib path I think

buxtronix, 2020-06-21:
Was this installed in your libraries folder? That would cause this error typically.

frizbeano, 2020-06-23:
Hi Buxtronix, No this was not installed into my library folder. I just downloaded and opened up the example. This could have been the way I set this up since I'm new to using it

buxtronix, 2020-06-25:
Thanks, yes it should be installed in your libraries folder.

closed 2020-06-25 by buxtronix