buxtronix/am43 #5: Add auto-discovery for Home Assistant

T-REX-XP, 2020-06-28:
hi. thank you for contributing. Your project very useful. Can you please add auto-discovery function??
here is the example of  implementation: [](https://github.com/binsentsu/am43-ctrl)

thanks

buxtronix, 2020-07-02:
FYI, this should be available shortly, I've been putting some final tweaks to auto discovery.

buxtronix, 2020-07-02:
Support for this has been pushed to the repo, please give it a try and report back.

MrBigleg, 2020-07-02:
working for me

buxtronix, 2020-07-03:
great!

closed 2020-07-03 by buxtronix

jascdk, 2020-08-29:
Hi Bus:) BIG THANKS for your hot-damn-cool code! WORKS GREAT - besides the Auto Discovery. I can't get it to work. I have it enabled in the config and HA but nothing shows up. I don't know how to troubleshoot it:(