buxtronix/am43 #6: Close then disconnect.

MrBigleg, 2020-07-01:
Firstly, thanks for this it is a great addition to my home automation.

I am having a few issues though.

Disconnects. Wich I've been following the other comment on.

This issue though is, if the blind is left in the fully open position. It will eventually close on its own and then disconnect, taking some time to become available again.

I am unsure on how to supply logs etc as I'm not too advanced in Arduino, all my devices are either ESPHome or Tasmota.

Is there anything I needed to do in the setup that I may have missed? One thing I noticed is that my open/closed positions reported are the opposite of your readme. I have tried swapping them but it just reports open when closed and it still closes itself after a random amount of time with nothing I can see triggering it.

I do have another ESP module in the same room it's used for Xiaomi plant and climate sensors.

I have also set the MAC of the blind in the config file.

I have also tried a second board. same problem.

Apologies if this is the wrong way/format to respond but it seems the only way to reach out to you.

Thanks again

buxtronix, 2020-07-02:
> My issue though is, if the blind is left in the fully open position. It will eventually close on its own and then disconnect, taking some time to become available again.

I can't imagine anything closing it in the code. Only two things come to mind:

- You have something sending a close command on the MQTT topic.
- You have the timers set and enabled in the Blind Engine app.

> I am unsure on how to supply logs etc as I'm not too advanced in Arduino, all my devices are either ESPHome or Tasmota.

Can you open the serial monitor in arduino and report what you see there?

> One thing I noticed is that my open/closed positions reported are the opposite to your readme. 

Perhaps HA does not behave well when the positions are inverted from the code? I'll be uploading a few changes shortly, it will have settings to invert the position numbers.

buxtronix, 2020-07-02:
I've played a bit with the positions here, the only thing that would give the swapped open/closed positions is if you have the motor direction set wrong. Open the Blinds Engine app and try changing the 'blind direction setting'.

MrBigleg, 2020-07-02:
Thanks for your reply.

I decided to reset everything and start again. 

1. Swap the motor around, and set the direction in the app to match the inverted numbers.
2. Connected and it seemed well it, then slightly opened the blinds, to which I found this via serial...

MQTT [am43/02:ba:58:19:2c:d3/set_position]2: 98
Addr: 02:ba:58:19:2c:d3 Cmd: set_position
[ 02:ba:58:19:2c:d3] AM43 Send: 00FF00009A0D0162F4
[02:ba:58:19:2c:d3] Got position: 99
BLE scan complete.

3. I found a command in Node-Red matching it, so maybe I made a mistake in there and it was sending a command as you suggested. So I removed it.

4. I then looked at this github and noticed you'd made some changes, so I decided to update everything, I also noticed this recommended change, so I amended the file (was it there before?)...

 m_scan_params.scan_duplicate     = BLE_SCAN_DUPLICATE_DISABLE;

5. I then removed the manual entity and sensors and I'm now testing but maybe Node-Red and an error from myself was the issue.

I will report back after I've tested for a while.

Thanks again

MrBigleg, 2020-07-02:
I think this issue can be closed, it seems a rogue Node-Red command was my issue coupled with the way it was set up.

The system is now constantly disconnecting, so I will comment on the other thread

closed 2020-07-02 by MrBigleg