buxtronix/am43 #8: Position topic broken if device name is used

Kite-Smurf, 2020-07-06:
I have just discovered that the position topic is broken, since the HA auto discovery function, if device name is used in the sensor config.

This does not update:
- platform: mqtt
  name: "Kitchen N Pos"
  state_topic: "am43/Kitchen North/Position"
  availability_topic: "am43/Kitchen North/available"
  unit_of_measurement: "%"

This works properly:

- platform: mqtt
  name: "Kitchen W Pos"
  state_topic: "am43/02874f1da497/position"
  availability_topic: "am43/Kitchen West/available"
  unit_of_measurement: "%"

buxtronix, 2020-07-07:
When auto-discovery is enabled, the topics use the mac address. This is to ensure uniqueness.

You should either disable auto-discovery in config.h, or remove the entries from configuration.yaml in HA.

Kite-Smurf, 2020-07-07:
That is what I did when auto-discovery was introduced. I had auto-discovery in config.h enabled and all entries removed from configuration.yaml in HA, however the position sensor was not populating. Then I tried setting up the position sensors manually.
The rest van de entities are created properly.

This is how it looks in the Developer Tools of HA:
[image]

buxtronix, 2020-07-07:
I'd be interested in seeing the MQTT data being sent, particularly after a restart.

Both am43/# and homeassistant/# to see what it's sending for auto discovery.

The entity names should be the device name not mac (cover.kitchen_west for example)

Kite-Smurf, 2020-07-07:
I need to correct my previous message. this is the result of the auto discovery:

[image]

The device name is used in the topic not the MAC. However the position topic is missing.

buxtronix, 2020-07-07:
If auto-discovery is enabled, then the mac will be used in the topic. If auto-discovery is disabled, the topic will depend on the AM43_USE_NAME_FOR_TOPIC define.

Are you running from the latest commit?

Kite-Smurf, 2020-07-07:
Yes, I'using the latest commit as the covers, battery sensors, light sensors were discovered automatically, except for the position. Position sensors never showed up.
All (manually created) entities were removed from the covers.yaml and sensors.yaml.

This is what the MQTT explorer comes up with: 
(all these topics update regularly) 

02bd8eb058b1
available = online
position = 0
battery = 50
set = OPEN
light = 48
02cc5aa4e276
available = online
position = 0
battery = 67
set = OPEN
light = 48
02874f1da497
available = online
position = 1
set = OPEN
battery = 47
light = 48

These seem to to be "old" retained topics. Not sure why they show up, but not updating:

Kitchen West
position = 1
available = online
LWT = Online
battery = 96
position = 99
light = 51
available = online
Dining North
battery = 52
position = 0
light = 54
available = online
Kitchen North
battery = 93
position = 0
light = 54
available = online
02:bd:8e:b0:58:b1
available = offline
battery = 96
position = 0
light = 54
02:87:4f:1d:a4:97
battery = 91
position = 100
light = 58

I hope this helps.

buxtronix, 2020-07-10:
There wont be a separate position entity. This is contained within the cover itself. By clicking on the device name in the lovelace UI, the popup shows a slider for the position. HA doesnt show a separate entity for the position.

buxtronix, 2020-07-10:
Example:

[image]

Kite-Smurf, 2020-07-10:
Thanks for the info. I created the Position entities manually using the MAC in the topic and that seems to be working.
[image]

We can close this then.

closed 2020-07-10 by Kite-Smurf