buxtronix/am43 #10: after adding the third blind all offline

Sildenafil99, 2020-07-22:
I have been using your controller for a couple of weeks with two blinds and everything has worked perfectly, except for the battery consumption which seems excessive to me.
today I added another blind but I don't know why all three stopped working.
They are controlled by an esp 32 located in the middle about 3-4 meters away from each blind.
They are all unavaible on homeassistant.

What's the problem?

buxtronix, 2020-07-22:
Are you able to grab serial logs for this? Unfortunately I dont have a 3rd motor to reproduce.

Sildenafil99, 2020-07-23:
How can i grab the serial log?

buxtronix, 2020-07-23:
Plug the device in a PC and open the Arduino serial monitor.

Sildenafil99, 2020-07-23:
13:42:36.565 -> ets Jun  8 2016 00:22:57
13:42:36.565 -> 
13:42:36.565 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:42:36.565 -> configsip: 0, SPIWP:0xee
13:42:36.565 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:42:36.612 -> mode:DIO, clock div:1
13:42:36.612 -> load:0x3fff0018,len:4
13:42:36.612 -> load:0x3fff001c,len:1216
13:42:36.612 -> ho 0 tail 12 room 4
13:42:36.612 -> load:0x40078000,len:9720
13:42:36.612 -> ho 0 tail 12 room 4
13:42:36.612 -> load:0x40080400,len:6352
13:42:36.612 -> entry 0x400806b8
13:42:37.075 -> Starting Arduino BLE Client application...
13:42:38.054 -> Wifi connecting to: XXXXXXXXXXX ... 
13:42:38.147 -> [WiFi-event] event: 0
13:42:38.147 -> [WiFi-event] event: 2
13:42:38.380 -> [WiFi-event] event: 4
13:42:38.380 -> [WiFi-event] event: 7
13:42:38.380 -> WiFi connected
13:42:38.380 -> IP address: 192.168.1.191
13:42:38.615 -> Attempting MQTT connection...BLE Advertised Device found: Name: , Address: 1a:68:3f:fe:6f:f6, manufacturer data: 0600010920025dc5de6fec8cff2244bdbc418687e792c5f055d05d74e7
13:42:39.036 -> connected
13:42:39.967 -> BLE Advertised Device found: Name: Flower care, Address: c4:7c:8d:6a:67:20, serviceUUID: 0000fe95-0000-1000-8000-00805f9b34fb
13:42:40.751 -> BLE Advertised Device found: Name: oscurante sx, Address: 02:f4:ae:78:e3:46, manufacturer data: 0000, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
13:42:41.031 -> Attempting to connect to: 02:f4:ae:78:e3:46 ...[02:f4:ae:78:e3:46] Connected
13:42:41.639 ->  - Failed to connect.
13:42:41.778 -> BLE Advertised Device found: Name: , Address: 02:13:b0:1d:5a:89, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
13:42:42.153 -> Attempting to connect to: 02:13:b0:1d:5a:89 ...[02:13:b0:1d:5a:89] Connected
13:42:42.340 ->  - Connected to server
13:42:43.636 -> Connect took 1503ms
13:42:43.682 -> BLE Advertised Device found: Name: , Address: 1a:68:3f:fe:6f:f6, manufacturer data: 0600010920025dc5de6fec8cff2244bdbc418687e792c5f055d05d74e7
13:42:44.005 -> BLE Advertised Device found: Name: , Address: 02:cb:51:91:b6:29, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
13:42:44.146 -> [ 02:13:b0:1d:5a:89] AM43 Send: 00FF00009A170222B815
13:42:44.146 -> Attempting to connect to: 02:cb:51:91:b6:29 ...[ 02:13:b0:1d:5a:89] Pin ok
13:42:44.517 -> [02:cb:51:91:b6:29] Connected
13:42:44.517 ->  - Failed to connect.
13:42:44.517 -> BLE Advertised Device found: Name: , Address: 1a:68:3f:fe:6f:f6, manufacturer data: 0600010920025dc5de6fec8cff2244bdbc418687e792c5f055d05d74e7
13:42:44.842 -> BLE Advertised Device found: Name: , Address: 02:cb:51:91:b6:29, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
13:42:44.842 -> Ignoring advertising device ⸮4⸮?U, already present
13:42:45.217 -> BLE Advertised Device found: Name: , Address: 02:f4:ae:78:e3:46, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
13:42:45.500 -> Attempting to connect to: 02:f4:ae:78:e3:46 ...

Sildenafil99, 2020-07-23:
I think the problem is tho other ble device like miflora.
I can try to add mac address to DEVICE_ALLOWLIST

Sildenafil99, 2020-07-23:
I tried to add another esp, one in DEVICE_ALLOWLIST with two blinds and the other one with one blind, but they goes offline after a few minutes

buxtronix, 2020-07-23:
I can't see anything obvious in the logs. You may need to increase the log verbosity.

Tools -> Core Debug Level -> Debug

Sildenafil99, 2020-07-24:
`21:28:33.452 -> ets Jun  8 2016 00:22:57
21:28:33.452 -> 
21:28:33.452 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:28:33.452 -> configsip: 0, SPIWP:0xee
21:28:33.452 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:28:33.452 -> mode:DIO, clock div:1
21:28:33.452 -> load:0x3fff0018,len:4
21:28:33.452 -> load:0x3fff001c,len:1216
21:28:33.499 -> ho 0 tail 12 room 4
21:28:33.499 -> load:0x40078000,len:9720
21:28:33.499 -> ho 0 tail 12 room 4
21:28:33.499 -> load:0x40080400,len:6352
21:28:33.499 -> entry 0x400806b8
21:28:33.916 -> [W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
21:28:33.963 -> Starting Arduino BLE Client application...
21:28:34.932 -> Wifi connecting to: XXXXXXXXXX ... 
21:28:35.024 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
21:28:35.024 -> [WiFi-event] event: 0
21:28:35.024 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
21:28:35.024 -> [I][ArduinoOTA.cpp:130] begin(): OTA server at: esp32-a4cf1281e8d0.local:3232
21:28:35.302 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
21:28:35.302 -> [WiFi-event] event: 4
21:28:35.302 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
21:28:35.302 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.191, MASK: 255.255.255.0, GW: 192.168.1.1
21:28:35.302 -> 192.168.1.191
21:28:35.534 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffeeac0), owner: <N/A> for start
21:28:35.534 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffeeac0), owner: start
21:28:35.534 -> Attempting MQTT connection...[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -65
21:28:35.721 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 29, data: 060001092002c5990511a08e0db4ab218acd99cc800fd63e56d29ab77c
21:28:35.767 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 060001092002c5990511a08e0db4ab218acd99cc800fd63e56d29ab77c
21:28:35.767 -> BLE Advertised Device found: Name: , Address: 1c:71:38:96:74:e3, manufacturer data: 060001092002c5990511a08e0db4ab218acd99cc800fd63e56d29ab77c
21:28:35.953 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 1c:71:38:96:74:e3, already seen it.
21:28:36.046 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 1c:71:38:96:74:e3, already seen it.
21:28:36.092 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -81
21:28:36.092 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x02 (), length: 2, data: 50fe
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 12, data: 6f73637572616e7465207378
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: oscurante sx
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 2, data: 0000
21:28:36.185 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 0000
21:28:36.185 -> BLE Advertised Device found: Name: oscurante sx, Address: 02:f4:ae:78:e3:46, manufacturer data: 0000, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
21:28:36.185 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: clients (0x3ffba454), owner: <N/A> for clientsAll
21:28:36.185 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: clients (0x3ffba454), owner: clientsAll
21:28:36.185 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: clients (0x3ffba454), owner: <N/A> for clientInsert
21:28:36.231 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: clients (0x3ffba454), owner: clientInsert
21:28:36.787 -> connected
21:28:36.787 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: clients (0x3ffba454), owner: <N/A> for clientsAll
21:28:36.787 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: clients (0x3ffba454), owner: clientsAll
21:28:36.787 -> Attempting to connect to: 02:f4:ae:78:e3:46 ...[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 0, GATT role: client
21:28:36.787 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3ffee3ac), owner: <N/A> for connect
21:28:36.787 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3ffee3ac), owner: connect
21:28:36.833 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:36.833 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:36.833 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3ffef424), owner: <N/A> for connect
21:28:36.833 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3ffef424), owner: connect
21:28:38.172 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.172 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.219 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.219 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.219 -> [02:f4:ae:78:e3:46] Connected
[... cut ...]
21:28:46.257 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3fff1f70), owner: connect
21:28:46.257 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... Unknown
21:28:46.304 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... Unknown
21:28:46.304 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3fff3ca0), owner: <N/A> for connect
21:28:46.304 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3fff3ca0), owner: connect
`

Sildenafil99, 2020-07-24:
21:28:33.452 -> ets Jun  8 2016 00:22:57
21:28:33.452 -> 
21:28:33.452 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:28:33.452 -> configsip: 0, SPIWP:0xee
21:28:33.452 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:28:33.452 -> mode:DIO, clock div:1
21:28:33.452 -> load:0x3fff0018,len:4
21:28:33.452 -> load:0x3fff001c,len:1216
21:28:33.499 -> ho 0 tail 12 room 4
21:28:33.499 -> load:0x40078000,len:9720
21:28:33.499 -> ho 0 tail 12 room 4
21:28:33.499 -> load:0x40080400,len:6352
21:28:33.499 -> entry 0x400806b8
21:28:33.916 -> [W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
21:28:33.963 -> Starting Arduino BLE Client application...
21:28:34.932 -> Wifi connecting to: XXXXXXXX ... 
21:28:35.024 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
21:28:35.024 -> [WiFi-event] event: 0
21:28:35.024 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
21:28:35.024 -> [I][ArduinoOTA.cpp:130] begin(): OTA server at: esp32-a4cf1281e8d0.local:3232
21:28:35.302 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
21:28:35.302 -> [WiFi-event] event: 4
21:28:35.302 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
21:28:35.302 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.191, MASK: 255.255.255.0, GW: 192.168.1.1
21:28:35.302 -> 192.168.1.191
21:28:35.534 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffeeac0), owner: <N/A> for start
21:28:35.534 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffeeac0), owner: start
21:28:35.534 -> Attempting MQTT connection...[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -65
21:28:35.721 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 29, data: 060001092002c5990511a08e0db4ab218acd99cc800fd63e56d29ab77c
21:28:35.767 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 060001092002c5990511a08e0db4ab218acd99cc800fd63e56d29ab77c
21:28:35.767 -> BLE Advertised Device found: Name: , Address: 1c:71:38:96:74:e3, manufacturer data: 060001092002c5990511a08e0db4ab218acd99cc800fd63e56d29ab77c
21:28:35.953 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 1c:71:38:96:74:e3, already seen it.
21:28:36.046 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 1c:71:38:96:74:e3, already seen it.
21:28:36.092 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -81
21:28:36.092 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x02 (), length: 2, data: 50fe
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 12, data: 6f73637572616e7465207378
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: oscurante sx
21:28:36.138 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 2, data: 0000
21:28:36.185 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 0000
21:28:36.185 -> BLE Advertised Device found: Name: oscurante sx, Address: 02:f4:ae:78:e3:46, manufacturer data: 0000, serviceUUID: 0000fe50-0000-1000-8000-00805f9b34fb
21:28:36.185 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: clients (0x3ffba454), owner: <N/A> for clientsAll
21:28:36.185 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: clients (0x3ffba454), owner: clientsAll
21:28:36.185 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: clients (0x3ffba454), owner: <N/A> for clientInsert
21:28:36.231 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: clients (0x3ffba454), owner: clientInsert
21:28:36.787 -> connected
21:28:36.787 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: clients (0x3ffba454), owner: <N/A> for clientsAll
21:28:36.787 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: clients (0x3ffba454), owner: clientsAll
21:28:36.787 -> Attempting to connect to: 02:f4:ae:78:e3:46 ...[I][BLEDevice.cpp:593] addPeerDevice(): add conn_id: 0, GATT role: client
21:28:36.787 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegEvt (0x3ffee3ac), owner: <N/A> for connect
21:28:36.787 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3ffee3ac), owner: connect
21:28:36.833 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:36.833 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:36.833 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3ffef424), owner: <N/A> for connect
21:28:36.833 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3ffef424), owner: connect
21:28:38.172 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.172 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.219 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.219 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
21:28:38.219 -> [02:f4:ae:78:e3:46] Connected
[... cut ...]
21:28:46.257 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegEvt (0x3fff1f70), owner: connect
21:28:46.257 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... Unknown
21:28:46.304 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... Unknown
21:28:46.304 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3fff3ca0), owner: <N/A> for connect
21:28:46.304 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3fff3ca0), owner: connect

adotagarwal, 2020-07-29:
you should look at the other open issue, this is the exact same behavior noted there. You should try the fixes noted there

Sildenafil99, 2020-07-29:
What fix? I'm nota able with coding..

adotagarwal, 2020-08-03:
I'll post the changes in the other thread. For this thread, I am also experiencing a drop of all devices after a while, however, in reality it is just the WiFi that has stopped and is now responding AP_NOTFOUND 

I need to look into that separately.

adotagarwal, 2020-09-04:
I've been running it fairly stable after making some updates. You need to apply the code changes in the readme of this project, as well as the ones mentioned in the other thread in the submitted change list for the arduino-esp proj.

I made a few additional tweaks, which I am diff'ing and will submit soon

Sildenafil99, 2020-09-04:
I've allready applied the code changes in the readme of this project, but where i can find the other fix?
Thx

ukdavewood, 2020-09-05:
I've got a setup with 10xAM43's and 7xESP32s and have been getting lots of stability problems, with the version previous to the less frequent battery change message. Have tried lots of combinations of 1,2 and 3 blinds on the same ESP.   The hangs mostly seem to occur After the  'attempting to connect ....' message.  Last night I made a change to all of the controllers to introduce a WDI timeout of 6 seconds with a wdt_reset at the end of 'loop'.  

Since making this tweak I have done lots of physical, and MQTT switch turn on and turn offs and so far the WDI timeout seems to have recovered all failures to connect without hanging the controller.

The only problem I am now getting is that on one controller (With 2 blinds attached) most of the time, and some controllers occasionally the 'switch off' causes a reset (not a WDT one mostly), which causes the controller to turn back on.  After a couple of tries the switch offs seem to stick - so I can probably work around this with some home assistant scripting - would be nice though if the controllers remembered their on/off state across resets.

Will look for the arduino-esp changes mentioned above too and run some more tests today before submitting any proposed changes.

ibl129, 2020-09-10:
> I've got a setup with 10xAM43's and 7xESP32s and have been getting lots of stability problems, with the version previous to the less frequent battery change message. Have tried lots of combinations [...]

I am having the same issues. Looking forward to your proposed changes though.

ukdavewood, 2020-09-10:
All been working well now since I made the change with no irrecoverable crashes. Have stuck with only 2 AM43’s per ESP32 as they seem to reset themselves less.  Managed to write some home assistant automations using a binary switch to remember the desired state of the AM43 switch and then turn the Switches back off if they turn themselves off after a reset. 
Not sure if 6 seconds is optimal for the timeout.
Will try to work out later today how to upload the proposed changes.

from David Wood

ukdavewood, 2020-09-11:
Don't really know how to use Github properly - but have created a fork with my proposed changes in it to the ino and h file only in the example folder.
Changes proposed
1. Introduced optional WDT timeout - initially set to 6 seconds - but could probably be pushed up a bit to reduce the number of resets.
2. Made a few tweaks to debug messages
3. Changed the consts to include the MQTT_PREFIX define - not sure if this is needed - but it may have helped when I went to second ESP32
4. Added a few defines for individual motor mac addresses - to make it easier to switch around multiple esp32s
5. The versions also has OTA set off - didn't really test this with WDT - so it may well be ok.
The WDT timeout has solved all of my hanging problems - some of my ESP32's reset themselves now and again, but they always seem to recover.  
I had to introduce some home assistant automation to handle the fact that when the ESP32 resets itself it turns its switch back on if it was previously turned off.

I tried three different types of ESP32s
- ExpressIf ones with shielding and onboard antenna - these ones gave me the most trouble, 
- Wemos Lolin32 Lit boards without shielding - this seem to have a better range than the shielded ones and I have therefore found them more reliable
- ezsbc ones with external antenna's only - this seem to work even better

The changes I have proposed are not the exact version I have been testing - as my version didn't include the recent 'multiple battery messages suppression' change, but I have tried the proposed version out a few times today without any issues.
I have restricted myself to 2 motors per ESP32 as I found this more reliable - but 3 motors no longer cause hangs - but they do seem to result in more resets.

ibl129, 2020-09-11:
That sounds pretty awesome Dave! I want to try all this myself as well. Where can I get the files from?

ukdavewood, 2020-09-11:
Inl129, If you click on the number to the right of 'fork' you will hopefully see mine at the bottom.  Then just download the two updated ino and h example files. - if you click on where it says update just above the source code it shows you a summary of the proposed changes, 

Sorry buxtronix if I am not doing this the correct way - This is the first time for me using git/github. Thanks for the great work.

matda59, 2020-12-15:
Also started having issues when connected a 3rd AM43 Blind to the ESP32 that is controlling the 2 blinds.

matda59, 2020-12-23:
> 
> 
> Also started having issues when connected a 3rd AM43 Blind to the ESP32 that is controlling the 2 blinds.

@ukdavewood - outstanding. this revised version worked. i have 3 AM43 blind motors, and 2 ESP32's. so far its working with 2 AM43 blinds on 1 ESP, and the remaining AM43 on the other ESP32. i'll see if i can get them all working on the same ESP32 soon.

ukdavewood, 2020-12-24:
Great. I have had 10 x AM43's connected to 7 x ESP32s working  continuously since September without issue using the slightly tweaked version mentioned on 11th Sept. Have been using the switch off option (with some home assistant automation) and have to recharge motors about twice during this period. All but 2 of the blinds have been open and closed successfully every day during this period.
Max number of blinds I got to work is 2 per ESP32.

buxtronix, 2021-02-11:
If you're still having issues, please try the latest version, the readme tells you how to use the leaner NimBLE library.

closed 2021-02-11 by buxtronix