buxtronix/am43 #13: Light sensor readings

adotagarwal, 2020-07-30:
I am wondering if there's an issue with the code that reads the light sensor? It is picking the 5th byte and both the other projects I see interfacing with this device use the 4th byte. I did some testing with the sensor and a stupid bright flash light and the 4th byte seems to move closer to 0 in darkness and further away (max at 9, so far) using the flash light. 

For reference, a python client returns the 4th byte * 12.5 (no clue where they got this from) for %light, another project in nodejs just returns the 4th byte as is.

What do you think @buxtronix ?

adotagarwal, 2020-07-30:
Here's the node js project: https://github.com/binsentsu/am43-ctrl/blob/master/src/am43.js#L137
Here's the python one: https://github.com/TheBazeman/A-OK-AM43-Blinds-Drive/blob/master/AOK-AM43.py#L73

buxtronix, 2020-07-30:
Well spotted! Indeed I was using the wrong byte - explains the minimal variance in the value. I'll get this fixed.

buxtronix, 2020-07-31:
This fix is now pushed.

closed 2020-07-31 by buxtronix

adotagarwal, 2020-08-03:
Awesome, I'll check it out. Thanks. I also want to help you make some battery efficiency changes, would you be amenable to working on this together?

buxtronix, 2020-08-03:
Seems beneficial, yes. Did you have any specific changes in mind?