buxtronix/am43 #14: Error in sketch

balbulator, 2020-08-03:
Help me to understand

Arduino: 1.8.13 (Windows 7), Board:"NodeMCU-32S, 80MHz, 115200"

C:\Users\Sergio\Desktop\am43-master\examples\MQTTBlinds\MQTTBlinds.ino: In member function 'virtual void MyAM43Callbacks::onConnect(AM43Client*)':

MQTTBlinds:111:19: error: 'class PubSubClient' has no member named 'setBufferSize'

       this->mqtt->setBufferSize(512);

                   ^
exit status 1

'class PubSubClient' has no member named 'setBufferSize'

Am I missing something? Thanks in advance

buxtronix, 2020-08-03:
You need to use the latest version of PubSubClient, 2.8.

closed 2020-08-03 by buxtronix

balbulator, 2020-08-03:
Thank you very much, you brought me out of the tracknot!