buxtronix/am43 #16: MQTT overran with Battery messages

chierichetto, 2020-08-27:
My AM43 supposedly can't decide between 97 and 100 percent, and is reporting everything back to Home Assistant. 
Maybe add an interval between MQTT publishes to not push at every state change, if something like this happens?
My other one does not behave like this.
[image]

buxtronix, 2020-08-28:
Yes this has annoyed me also. I've pushed an update to the repo which limits messages to once every 30s.

closed 2020-08-28 by buxtronix

chierichetto, 2020-09-03:
Thanks for the quick fix. I also have some problems with the curtains becoming unavailable in HA after a while, but that's a separate issue i haven't been able to get any logs for yet. I'll open a new issue when i have isolated the fault. Maybe a reflash will fix it as well.