buxtronix/am43 #20: set CLOSE \ OPEN not working

matda59, 2021-02-06:
When i click on the close blind button it doesn't do anything for this particular blind, all others close and open. I can see that the command is going to the blind, but it doesn't close or open the blind - However changing the position of the blind works fine.

[image]

[image]

[image]

buxtronix, 2021-02-09:
This would imply that the device has not been setup with the 'Blinds Engine' app properly. Please verify that the limits, direction, etc are all as expected.

Otherwise if you can grab a log of the serial port while this is happening, that will help troubleshoot.

matda59, 2021-02-09:
Outstanding Buxtronix - that worked resetting the blind limits.

closed 2021-02-09 by matda59