buxtronix/am43 #21: Esphome feature request?

jascdk, 2021-02-27:
Hi @buxtronix 

first of all - thanks a lot for your work with these blind engines - it is a nice addition to our smarthome;) btw - have you ever concidered to move / port the project over to esphome ?

DaveTiff, 2021-02-27:
That would be really good.

buxtronix, 2021-02-27:
This is already in progress, I've got an open PR to enable ble client support:

https://github.com/esphome/esphome/pull/1177

With a follow-up one that then enables am43 support. You can check out my fork here:

https://github.com/buxtronix/esphome/tree/am43

jascdk, 2021-02-28:
Wow that sounds great! Is it something I can try out or is it not ready yet?

buxtronix, 2021-02-28:
Yes the fork works very well. It's been controlling a pair of my blinds daily for many months.

jascdk, 2021-02-28:
@buxtronix great 🤘 Sorry for asking a lot, but how do I use your fork ? Can I copy the sm43 libraries over in my existing esphome under HA?

buxtronix, 2021-03-01:
You would need the entire repo as some of the other parts of the esphome codebase are changed.

If you are using Hass.io, I think you can set the config options "esphome_version" to "buxtronix:am43" though I havent been able to try this myself.

jascdk, 2021-03-01:
Sounds like the Way to go @buxtronix ! I Will try - though om afraid it Will mess up my other configurations :/ maybe I will wait til your pr goes live in a new release . Any clue how long it will go before merging it int the base code ?

T-REX-XP, 2021-03-01:
> If you are using Hass.io, I think you can set the config options "esphome_version" to "buxtronix:am43" though

Thanks, it works. Do you have got a configuration example for the am43 gateway ??

buxtronix, 2021-03-01:
This was my bad, you found a bug :)

Please re-fetch and try again. (though i'm not sure how to force a re-fetch, perhaps flip 'esphome_version' to the previous, restart, then back to my fork.)

T-REX-XP, 2021-03-01:
> Please re-fetch and try again. (though i'm not sure how to force a re-fetch, perhaps flip 'esphome_version' to the previous, restart, then back to my fork.)

still have an issue((

```
Compiling /data/am43_gateway/.pioenvs/am43_gateway/src/esphome/components/am43_cover/am43_base.cpp.o
Compiling /data/am43_gateway/.pioenvs/am43_gateway/src/esphome/components/am43_cover/am43_cover.cpp.o
Compiling /data/am43_gateway/.pioenvs/am43_gateway/src/esphome/components/api/api_connection.cpp.o
Compiling /data/am43_gateway/.pioenvs/am43_gateway/src/esphome/components/api/api_pb2.cpp.o
src/esphome/components/am43_cover/am43_cover.cpp: In member function 'virtual void esphome::am43_cover::Am43Cover::loop()':
src/esphome/components/am43_cover/am43_cover.cpp:24:13: error: 'class esphome::am43_cover::Am43Cover' has no member named 'node_state_'
   if (this->node_state_ == espbt::ClientState::Established && !this->logged_in_) {
             ^
src/esphome/components/am43_cover/am43_cover.cpp:26:59: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if_'
     auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                           ^
src/esphome/components/am43_cover/am43_cover.cpp:26:85: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id_'
     auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                                                     ^
src/esphome/components/am43_cover/am43_cover.cpp: In member function 'virtual void esphome::am43_cover::Am43Cover::control(const esphome::cover::CoverCall&)':
src/esphome/components/am43_cover/am43_cover.cpp:46:13: error: 'class esphome::am43_cover::Am43Cover' has no member named 'node_state_'
   if (this->node_state_ != espbt::ClientState::Established) {
             ^
src/esphome/components/am43_cover/am43_cover.cpp:52:59: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if_'
     auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                           ^
src/esphome/components/am43_cover/am43_cover.cpp:52:85: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id_'
     auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                                                     ^
src/esphome/components/am43_cover/am43_cover.cpp:61:59: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if_'
     auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                           ^
src/esphome/components/am43_cover/am43_cover.cpp:61:85: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id_'
     auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                                                     ^
src/esphome/components/am43_cover/am43_cover.cpp: In member function 'virtual void esphome::am43_cover::Am43Cover::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/am43_cover/am43_cover.cpp:82:33: error: 'class esphome::ble_client::BLECharacteristic' has no member named 'handle_'
       this->char_handle_ = chr->handle_;
                                 ^
src/esphome/components/am43_cover/am43_cover.cpp:84:70: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if_'
       auto status = esp_ble_gattc_register_for_notify(this->parent_->gattc_if_, this->parent_->remote_bda_,
                                                                      ^
src/esphome/components/am43_cover/am43_cover.cpp:84:96: error: 'class esphome::ble_client::BLEClient' has no member named 'remote_bda_'
       auto status = esp_ble_gattc_register_for_notify(this->parent_->gattc_if_, this->parent_->remote_bda_,
                                                                                                ^
src/esphome/components/am43_cover/am43_cover.cpp:85:60: error: 'class esphome::ble_client::BLECharacteristic' has no member named 'handle_'
                                                       chr->handle_);
                                                            ^
src/esphome/components/am43_cover/am43_cover.cpp:92:13: error: 'class esphome::am43_cover::Am43Cover' has no member named 'node_state_'
       this->node_state_ = espbt::ClientState::Established;
             ^
src/esphome/components/am43_cover/am43_cover.cpp:110:65: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if_'
           auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                                 ^
src/esphome/components/am43_cover/am43_cover.cpp:110:91: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id_'
           auto status = esp_ble_gattc_write_char(this->parent_->gattc_if_, this->parent_->conn_id_,
                                                                                           ^
*** [/data/am43_gateway/.pioenvs/am43_gateway/src/esphome/components/am43_cover/am43_cover.cpp.o] Error 1
========================= [FAILED] Took 10.58 seconds =========================
```

buxtronix, 2021-03-01:
Ah yes, this is a bit of a mess right now after some updates to the parent branch. I'll get it sorted and update today.

buxtronix, 2021-03-02:
It should be good now.

T-REX-XP, 2021-03-02:
> It should be good now.

After the flashing FW I have received an error - 

The firmware has been successfully compiled, but for some reason, I can't flash it via ESPHome, so I just tried to burn firmware.bin via nodeMCU Flasher. 

```
st:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
```

buxtronix, 2021-03-03:
This is my config:

```
esp32_ble_tracker:

ble_client:
  - mac_address: 02:4D:FB:F0:2B:2E
    id: art_left
  - mac_address: 02:69:32:F0:A5:1D
    id: art_right

cover:
  - platform: am43_cover
    name: "ArtLeft Cover"
    id: art_left_cover
    ble_client_id: art_left
  - platform: am43_cover
    name: "ArtRight Cover"
    id: art_right_cover
    ble_client_id: art_right

sensor:
  - platform: am43
    ble_client_id: art_left
    battery_level:
      name: "AM43 Left battery"
    illuminance:
      name: "AM43 Left light"
    update_interval: 120s
  - platform: am43
    ble_client_id: art_right
    battery_level:
      name: "AM43 Right battery"
    illuminance:
      name: "AM43 Right light"
    update_interval: 120s
  - platform: uptime
    name: Uptime Sensor

switch:
  - platform: restart
    name: "Device Restart"
  - platform: ble_client
    name: "Left AM43 On"
    ble_client_id: art_left
  - platform: ble_client
    name: "Right AM43 On"
    ble_client_id: art_right
    id: right_on
```

belenkiy-lab, 2021-03-07:
Hi 
First of all - Thanks for your great work.

I try your config with my am43.  I have two - left and right :) but only one works at a time

if I turn on both curtains, only one will work.
if I turn off any of them, then the second will work correctly.
the one that does not work gives an error - [am43_cover:079]: [ArtLeft Cover] No control service found at device, not an AM43..?[0m

below is the log after reboot. I use your config, the only other is mac

[https://pastebin.com/0LwmKjtt](url)

jascdk, 2021-03-08:
> Hi
> First of all - Thanks for your great work.
>
> I try your config with my am43. I have two - left and right :) but only one works at a time
>
> if I turn on both curtains, only one will work.
> if I turn [...]

I have the same problem with my 2 motors :(

JochenKlenk, 2021-03-11:
Hi,

I also want to thank you. 
I get the firmware on the device and it is running. But I not can connect to the AM43. At the app it is possible. I of course also delete the connection to the app before I try to access over the ESP32. Do you saw that issue before?

[14:04:36][D][esp32_ble_tracker:180]: Starting scan...
[14:04:41][D][ble_client:045]: Found device at MAC address [02:81:B1:08:0C:E7]
[14:04:41][I][ble_client:083]: Attempting BLE connection to 02:81:b1:08:0c:e7
[14:04:42][I][ble_client:140]: [02:81:b1:08:0c:e7] ESP_GATTC_DISCONNECT_EVT
[14:04:42][D][sensor:092]: 'AM43 Blind1 battery': Sending state nan % with 0 decimals of accuracy
[14:04:42][D][sensor:092]: 'AM43 Blind1 light': Sending state nan % with 0 decimals of accuracy
[14:04:42][I][ble_client:113]: [02:81:b1:08:0c:e7] ESP_GATTC_OPEN_EVT
[14:04:42][W][ble_client:115]: connect to 02:81:b1:08:0c:e7 failed, status=133
[14:04:42][D][esp32_ble_tracker:180]: Starting scan...
[14:04:52][W][am43:090]: [02:81:b1:08:0c:e7] Cannot poll, not connected
[14:05:10][D][sensor:092]: 'Uptime Sensor': Sending state 1184.43005 s with 0 decimals of accuracy

Thanks

JochenKlenk, 2021-03-15:
Anybody have an idea on post above?

Thanks

mj1985, 2021-03-17:
@JochenKlenk I have the same problem, I did manage to connect one of the device, after uploading via usb-cable and not OTA.

 But noticed with two am43-devices only one gets connected, the first gets connected (I see the pin-message) and the second the pin-message is missing. I also did a reset and everytime I upload via usb-cable or ota another AM43 gets connected first. The first connected device is perfectly working and the second not.

bds89, 2021-03-28:
Еhanks for the blind controller
I use your config:
```
sensor:
  - platform: am43
    ble_client_id: art_left
    battery_level:
      name: "Kitchen_left_blind battery"
    illuminance:
      name: "Kitchen_left_blind light"
    update_interval: 120s
```
But ESPHome gives error:
Platform not found: sensor.am43
Does anyone know how to solve? 
At the same time the am43_cover and ble_client components is connected and works great.

Kitser, 2021-04-26:
> Thanks for the blind controller
> I use your config:
> 
> ```
> sensor:
>   - platform: am43
>     ble_client_id: art_left
>     battery_level:
>       name: "Kitchen_left_blind battery"
>     illuminance:
>       name: "Kitchen_left_blind light"
>     update_interval: 120s
> ```
> 
> But ESPHome gives error: Platform not found: sensor.am43
> Does anyone know how to solve?
> At the same time the am43_cover and ble_client components is connected and works great.

I got the same problem.

But other than that it works great! Much easier than the original way of this repo. 
Thanks man!

Wolbyworld, 2021-04-30:
Hopefully a quick one, when using the fork in HA with the version: "buxtronix:am43" I get the following error:

`esphome requires Python '>=3.7,<4.0' but the running Python is 3.6.9`

My HA is running 3.8.7

JochenKlenk, 2021-05-03:
Hi,

same issue for me.

esphome requires Python '>=3.7,<4.0' but the running Python is 3.6.9

 System: Home Assistant OS 5.13  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2021.5.0b6
 Home Assistant Supervisor: 2021.04.3

Thanks

DaveTiff, 2021-05-04:
Is there a write up on how to use / setup the AM43 on a ESP32 using ESPHOME as my version of ESPHOME 1.17 (updated today) does recognize AM43 as a platform?

Regards Dave

pixeldublu, 2021-05-08:
@buxtronix 

Not sure if something was changed recently but today i tried to setup and it fails with for the sensor part:

```
Compiling /data/esp32/.pioenvs/esp32/src/esphome/components/am43/am43.cpp.o
In file included from src/esphome/components/am43_cover/am43_base.h:3:0,
                 from src/esphome/components/am43/am43.h:7,
                 from src/esphome/components/am43/am43.cpp:1:
src/esphome/components/am43/am43.cpp: In member function 'virtual void esphome::am43::Am43::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/am43/am43.cpp:46:91: error: 'class esphome::am43::Am43' has no member named 'get_name'
           ESP_LOGE(TAG, "[%s] Detected a Tuya AM43 which is not supported, sorry.", this->get_name().c_str());
                                                                                           ^
src/esphome/core/log.h:119:89: note: in definition of macro 'esph_log_e'
   esp_log_printf_(ESPHOME_LOG_LEVEL_ERROR, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^
src/esphome/components/am43/am43.cpp:46:11: note: in expansion of macro 'ESP_LOGE'
           ESP_LOGE(TAG, "[%s] Detected a Tuya AM43 which is not supported, sorry.", this->get_name().c_str());
           ^
src/esphome/components/am43/am43.cpp:48:90: error: 'class esphome::am43::Am43' has no member named 'get_name'
           ESP_LOGE(TAG, "[%s] No control service found at device, not an AM43..?", this->get_name().c_str());
                                                                                          ^
src/esphome/core/log.h:119:89: note: in definition of macro 'esph_log_e'
   esp_log_printf_(ESPHOME_LOG_LEVEL_ERROR, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^
src/esphome/components/am43/am43.cpp:48:11: note: in expansion of macro 'ESP_LOGE'
           ESP_LOGE(TAG, "[%s] No control service found at device, not an AM43..?", this->get_name().c_str());
           ^
Compiling /data/esp32/.pioenvs/esp32/src/main.cpp.o
*** [/data/esp32/.pioenvs/esp32/src/esphome/components/am43/am43.cpp.o] Error 1
```

TimSoethout, 2021-05-19:
Cool! It looks like https://github.com/esphome/esphome/pull/1177 is merged. 

What is now the easiest way to use your library? Vanilla esphome with this code in a local directory?

jascdk, 2021-05-21:
When does it come in official ESPhome release? Is it stable now - I tried it before, with two blind engines - only one did I get to work - they did not work together:(

balbulator, 2021-05-23:
> Спасибо за слепой контроллер,
> использую ваш конфиг:
> 
> ```
> sensor:
>   - platform: am43
>     ble_client_id: art_left
>     battery_level:
>       name: "Kitchen_left_blind battery"
>     illuminance:
>       name: "Kitchen_left_blind light"
>     update_interval: 120s
> ```
> 
> Но ESPHome выдает ошибку: Платформа не найдена: sensor.am43
> Кто-нибудь знает как решить?
> При этом компоненты am43_cover и ble_client подключены и отлично работают.

I also have an error
```
Platform not found: cover.am43_cover
Platform not found:  sensor.am43
```

Sildenafil99, 2021-06-06:
Hi, 
I saw that the pr was merged.
What is the working configuration?

Thx

balbulator, 2021-06-19:
ESPHome
Current version: 1.19.1
The blinds never worked

balbulator, 2021-07-14:
> This is my config:
>
> ```
> esp32_ble_tracker:
>
> ble_client:
>   - mac_address: 02:4D:FB:F0:2B:2E
>     id: art_left
>   - mac_address: 02:69:32:F0:A5:1D
>     id: art_right
>
> cover:
>   - platform: am43_cover [...]

I was able to overcome this problem, in this configuration I replaced 
>   - platform: am43_cover
on the
>   - platform: am43
the error disappeared and I was able to compile
Also important in -> Supervisor -> ESPHome -> Configuration -> 
esphome_version - buxtronix:am43
certfile - fullchain.pem
keyfile - privkey.pem

Glory buxtronix!

belenkiy-lab, 2021-07-15:
another way to use External Components
https://esphome.io/components/external_components.html

add to your config:
```
external_components:
  - source:
      type: git
      url: https://github.com/buxtronix/esphome.git
      ref: am43
    components: [ am43 ]
```

jascdk, 2021-07-22:
When using external components as mentioned I get the following? 

AM I doing something wrong?

``````
INFO Reading configuration /config/esphome/blinds.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/blinds
Processing blinds (board: esp32dev; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/blinds/.pioenvs/blinds/src/main.cpp.o
Compiling /data/blinds/.pioenvs/blinds/lib789/WiFi/WiFiClient.cpp.o
Compiling /data/blinds/.pioenvs/blinds/lib789/WiFi/WiFiGeneric.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:205:30: error: invalid new-expression of abstract class type 'esphome::am43::Am43'
   am43_am43 = new am43::Am43();
                              ^
In file included from src/esphome.h:2:0,
                 from src/main.cpp:3:
src/esphome/components/am43/am43.h:18:7: note:   because the following virtual functions are pure within 'esphome::am43::Am43':
 class Am43 : public esphome::ble_client::BLEClientNode, public PollingComponent {
       ^
In file included from src/esphome/components/am43/am43.h:4:0,
                 from src/esphome.h:2,
                 from src/main.cpp:3:
src/esphome/components/ble_client/ble_client.h:28:16: note: 	virtual void esphome::ble_client::BLEClientNode::loop()
   virtual void loop() = 0;
                ^
src/main.cpp:240:32: error: invalid new-expression of abstract class type 'esphome::am43::Am43'
   am43_am43_2 = new am43::Am43();
                                ^
*** [/data/blinds/.pioenvs/blinds/src/main.cpp.o] Error 1
========================= [FAILED] Took 13.52 seconds =========================
``````

buxtronix, 2021-07-23:
I have converted this into a repo supporting the new External Components feature of ESPHome. Please see https://github.com/buxtronix/esphome-am43

closed 2021-07-25 by jascdk

danielbrunt57, 2022-07-09:
Should anyone encounter **Platform not found:  sensor.am43** like I have been battling, check your `/config/esphome/.esphome/external_components/` for folders referencing `am43` and remove the entire folder(s).  Then restart the ESPHome add-on.  I just did that and now my am43-controller.yam,l is fine.