buxtronix/am43 #22: AM43 tuya

hanssipi, 2021-03-14:
Good job, buxtronix,

I bought a am43 from the tuya brand. Also a BT device but does not work with the blind engine app. the app can't find it. The device does advertises its mac.
Your sketch on my ESP32 runs and scans for the mac in the list but also never finds it. I can see other devices.
The smart tuya app works. I can see the mac of the device in the tuya app. In the tuya app there was no need to enter a pin so there is a difference.
Any ideas are welcome

...
Attempting MQTT connection...
connected
BLE Advertised Device found: Name: MUS000000, Address: aa:5c:8a:72:8e:c2, manufacturer data: 590001020304
BLE Advertised Device found: Name: [TV] Samsung 5 Series (49), Address: 14:cc:6e:cb:e4:0d, manufacturer data: 7500420401404f14bb6ecbe40d16bb6e50e40c30000000000000, txPower: 9
BLE scan complete.
Up for 60s
BLE scan complete.
...

Best Regards

T-REX-XP, 2021-03-14:
Can you send me a link for this item? I'm not sure that your devices  based on Bluetooth. Maybe zigbee version. You can also try to find ble devices via the phone just find ble scanner app in playstore.

hanssipi, 2021-03-14:
> Can you send me a link for this item? I'm not sure that your devices based on Bluetooth. Maybe zigbee version. You can also try to find ble devices via the phone just find ble scanner app in playstore.

Shure,
https://www.amazon.nl/Dan-Dre-Motorized-Shutter-Bluetooth/dp/B08TWRX36L/

hanssipi, 2021-03-14:
> ble scanner app

Finds the device.

andia89, 2021-04-07:
I have the same issue with a tuya AM34 device....
Problem is apparently that their protocol is just different (https://github.com/TuyaInc/tuya_ble_sdk)  as the service UUID seems to be 1910 and characteristics 2b11, but also the message that needs to be send is different than for "conventional" am34 devices

buxtronix, 2021-04-08:
Ah that would explain it then. So I cannot support the Tuya ones alas (unless someone wants to write support and send a PR?).

In the meantime I'll have to update the README to explicitly warn about these ones.

dhutchison, 2021-09-01:
@buxtronix did you reverse engineer the original AM43s to make this or was there a pretty good protocol understanding already available? 

I've never touched cpp and last time I touched c was well over a decade ago but I do have one of these tuya ble devices and keen to help reverse engineering if I can

buxtronix, 2021-09-02:
I didn't reverse engineer from scratch. I referenced a lot of existing code such as https://github.com/binsentsu/am43-ctrl but once I had the basics from there I also reverse engineered some more details.

dhutchison, 2021-09-03:
There seems to be very little about these new ones out there. 

I’ve got a BLE sniffer on the way, and already have tuya and original versions of the controller so I’ll probably be having fun with Wireshark next week and see what I can figure out. (Didn’t even notice it was tuya till after I’d installed it and tried to set this part up - this was meant to be an easy automation with the same hardware as before!)

nordeep, 2021-10-06:
> There seems to be very little about these new ones out there.
>
> I’ve got a BLE sniffer on the way, and already have tuya and original versions of the controller so I’ll probably be having fun with [...]

About sniff some BLE traffic. Could you try to sniff BLE packet which move original motor to one "small" step? For now, as I see, the smallest step is 1%. But by Blind App we can get one "small" step.

samlroberts, 2021-11-11:
Has anyone had any luck with this, especially sniffing out the packets? I did the same thing as @dhutchison and bought and installed the wrong ones, and was trying to make them work.

Kitser, 2022-06-24:
I already integrated AM43 with Home Assistant at a friend's house. So I bought one myself. Unfortunately it's a Smart Life version 😢
As far as I know there is no way to integrate it with Home Assistant. Or is there any news about this?

danielbrunt57, 2022-06-28:
I finally got around to doing this for myself and was successful.  These work...
[Moes DIY Smart Motorized Chain Roller Blinds Shade Shutter Drive Motor Powered By Charger Bluetooth APP Control](https://www.aliexpress.com/item/4000410549944.html?spm=a2g0o.productlist.0.0.9ddf17bdnVZbpS&algo_pvid=1b9c1a88-fdc0-442f-bb2e-17c5594f327b&algo_exp_id=1b9c1a88-fdc0-442f-bb2e-17c5594f327b-23&pdp_ext_f=%7B%22sku_id%22%3A%2210000013762796561%22%7D&pdp_npi=2%40dis%21CAD%21%2152.73%21%21%21%21%21%402101e9d016563883497394352e66bc%2110000013762796561%21sea)