buxtronix/am43 #25: HCI packet count mismatch

CountParadox, 2021-04-11:
I am getting the following error when attempting to use this to control my blinds.

`
Attempting MQTT connection...
connected
BLE Advertised Device found: Name: Guest Room, Address: 02:0f:95:ef:64:99, manufacturer data: 0000, serviceUUID: 0xfe50
BLE Advertised Device found: Name: VHM-ble, Address: 00:00:00:34:33:33, serviceUUID: 0xfff0
BLE scan complete.
Attempting to connect to: 02:0f:95:ef:64:99 ...lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
 - Failed to connect.
`

From my reading it looks like something to do with an abrupt disconnect / unclosed session, but I am not sure how to resolve it..

bibarrav, 2021-07-23:
I got the same issue, using the latest esp32 core version 2.0.0-rc1, and after downgrade to 1.0.6 it's works fine.

Emilef22, 2022-07-31:
Same issue with the 1.0.6, do you solve it? Thanks