buxtronix/am43 #26: Enable Bluetooth, can take 15 secs to 4 min to connect?

DaveTiff, 2021-05-03:
Hi 

I am using the recommended automation to first enable bluetooth wait x mins and then open curtains, but the time can vary before in connects, anything from 15 seconds to 4 minutes. Is this normal, anything I should look for.

I am using the NimBLE.

Regards Dave

buxtronix, 2021-05-03:
Usually this would suggest a poor signal between the AM43 and the ESP32. It might be worth bringing them closer or otherwise checking various causes of poor signal. In a couple of cases I have noted that the ESP32 has a tiny 0-ohm resistor thats set to use an (unconnected) external antenna.

DaveTiff, 2021-05-03:
Moved closer, no change, connected ESP back to my PC Serial Port Monitor, I get this

`BLE scan complete.
Attempting to connect to: 02:08:e8:b8:24:93 ... - Failed to connect.
Up for 240s
BLE scan complete.
Attempting to connect to: 02:08:e8:b8:24:93 ... - Failed to connect.
Up for 300s
BLE Advertised Device found: Name: Office window , Address: 02:08:e8:b8:24:93, manufacturer data: 0000, serviceUUID: 0xfe50
BLE scan complete.
Attempting to connect to: 02:08:e8:b8:24:93 ... - Failed to connect.
Up for 360s
BLE scan complete.
`

DaveTiff, 2021-05-03:
Think it must be a distance thing, moved it to about 250mm away and it seems to connect ok, is the ESP32 BL that bad?

buxtronix, 2021-05-04:
This link explains the likely issue (I had it also):

https://marksbench.com/electronics/esp32-cam-antenna-workaround/

DaveTiff, 2021-05-04:
Assume that the Blutooth and Wi-Fi share the same aerial?

DaveTiff, 2021-05-04:
Just flashed that same ESP32 with some ESPHOME code to show Wi-Fi Strength, seems to be the same level as other ESP devices I have.

So I assume the on board Antenna must be connected. 

Regards, Dave

buxtronix, 2021-05-05:
Possibly yes. I'm going to still blame it on your esp32 hardware - I easily get 5-10m range to my AM43s.

DaveTiff, 2021-05-06:
I will order another ESP32.

I would like to get it running from ESPHOME do you know when AM43 will be added as a device?

Regards Dave

DaveTiff, 2021-05-12:
Hi have now received another ESP32, and confirm that the range has improved to around 5m.

Thanks

Just need to get this installed into ESPHOME?

Regards, Dave

closed 2021-05-13 by DaveTiff