buxtronix/am43 #27: Battery, Light, not being pushed into MQTT

stratus-ss, 2021-05-21:
I have been using this project for several months now and I really appreciate the work. Recently the ESP32 has stopped pushing data into the MQTT server.

If I connect to the topic like so:

```
mosquitto_sub -h 192.168.9.160 -v -t projector_blind/# -u 'user -P 'password'
projector_blind/LWT Online
projector_blind/enabled ON
projector_blind/<mac add>/available online
```

However, I never see the battery or light being published. I also have to do this:

```
projector_blind/all/set -m OPEN
```

In order to interact with the blinds (as oppposed to using the mac address). Here is my config

```
// Your Wifi SSID.
#define WIFI_SSID "WifeyWifi"
 
// Wifi password.
#define WIFI_PASSWORD "happyhappy"
 
// MQTT server details.
#define MQTT_ADDRESS "192.168.9.160"
#define MQTT_USERNAME "user"
#define MQTT_PASSWORD "password"
 
// WDT Timeout (in seconds) - comment out if not required
#define WDT_TIMEOUT 6
 

#define AM43_USE_NAME_FOR_TOPIC
 
#define MQTT_TOPIC_PREFIX "projector_blinds"
 
#define AM43_ENABLE_MQTT_DISCOVERY
 
#define AM43_MQTT_DEVICE_CLASS "shade"
 
#define AM43_PIN 8888
 
#define SHADE_1 "02:A1:BD:A2:CD:80"
 
#define DEVICE_ALLOWLIST ""

#define ENABLE_ARDUINO_OTA
 
// LEAVE BELOW UNTOUCHED
#ifdef AM43_ENABLE_MQTT_DISCOVERY
#undef AM43_USE_NAME_FOR_TOPIC
#endif
 
#define BLE_MAX_CONN 3
```

any pointers would be helpful

DaveTiff, 2021-05-22:
What version of Mosquitto broker are you running?

stratus-ss, 2021-05-22:
```
Mosquitto broker
Current version: 6.0.1
```

DaveTiff, 2021-05-22:
Has this stopped working since the update, I am having issue not sure what the problem is, I have gone back to 5.1.1.

Just a thought!

stratus-ss, 2021-05-22:
I'll try rolling back and see what happens

stratus-ss, 2021-05-22:
So rolling back the broker indeed fixed the problem. I have disabled the auto-update. I wonder what changed in the broker that prevents this from functioning as expected

frizbeano, 2021-05-22:
I would like to add that I also have the issue

balbulator, 2021-05-22:
how to rollback the broker version?

stratus-ss, 2021-05-22:
I have snapshots going back a few months. I rolled back to one mid-april which had the previous version of the broker

DaveTiff, 2021-05-22:
Hi, I did the same, used a Snapshot and just checked MQTT

balbulator, 2021-05-22:
MQTT settings haven't changed?

CaptInsano, 2021-05-22:
Also happening here.
HomeAssistant 2021.5.5
Mosquitto: 6.0.1

All other MQTT devices (mainly Tasmota) all still working perfectly

MrBigleg, 2021-05-23:
+1 I have gone back to 5.1.1. All good again

closed 2023-01-05 by stratus-ss