buxtronix/am43 #28: Updating Mosquitto broker to 6.0.1 stops AM43 from working

DaveTiff, 2021-05-22:
Hi, updated my Mosquitto broker from 5.1.1 to 6.0.1 stops AM43 from operating. 

The Blue Tooth Enable works, but the components on HA dashboard remain Greyed Out. Restoring MQTT back to 5.1.1 cures this problem.

Regards, Dave

DaveTiff, 2021-05-22:
Now not sure it is that, as it has now also stopped working with 5.1.1. Restarting Mosquitto broker allows it to work again.

It has been working fine for over a week, Now running HA 2021.5.5

Anyone else having problems. 

Regards, Dave

DaveTiff, 2021-05-22:
Another person is now having trouble with Mosquitto broker 6.0.1 and AM43.

https://github.com/buxtronix/am43/issues/27#issuecomment-846403962

CaptInsano, 2021-05-22:
Also happening here.
HomeAssistant 2021.5.5
Mosquitto: 6.0.1

All other MQTT devices (mainly Tasmota) all still working perfectly

Kite-Smurf, 2021-05-23:
Hi, The exact same thing happens with my system too:
updated my Mosquitto broker from 5.1.1 to 6.0.1 stops AM43 from operating.

The Blue Tooth Enable works, but the components on HA dashboard remain Greyed Out. Restoring MQTT back to 5.1.1 cures this problem.

Regards,
Otto

balbulator, 2021-05-23:
added a question in the mosquito thread
https://github.com/home-assistant/addons/issues/2030

belenkiy-lab, 2021-05-23:
I have several curtains at home.
half are connected via this integration, half are connected via test version in esphome
after updating mqtt those connected via esphome remained working. the rest stopped working like everyone else.

balbulator, 2021-05-24:
> I have several curtains at home.
> half are connected via this integration, half are connected via test version in esphome
> after updating mqtt those connected via esphome remained working. the rest stopped working like everyone else.

tell me, how exactly did you connect the AM43 to ESPhome in the test?

belenkiy-lab, 2021-05-24:
It worked for me if I specified "buxtronix: am43" in the Esphome version. After that, everything worked. It was in early March. When I tried to do similar actions in April, I get the same error as yours.
therefore, I started new curtains not through esphome, but through native integration from buxtronix. I decided it was easier than trying to figure it out.
I'm waiting for this integration to be in the esphome release to use it fully and monotonously at home.
-------------------------------------------------------------------
У меня это работало если указать в версии Esphome "buxtronix:am43". После этого все работало. Это было в в начале марта. Когда я пробовал в апреле сделать аналогичные действия, я получаю ошибку, такую же как и у тебя.
поэтому новые шторы я завел не через esphome а через родную интеграцию от buxtronix. Решил что это проще, чем разбираться. 
Жду когда данная интеграция будет в релизе esphome что бы использовать ее полноценно и однообразно у себя.

balbulator, 2021-05-24:
none of the options work at the moment

belenkiy-lab, 2021-05-24:
Use mqtt version 5.1.1 and this integration

balbulator, 2021-05-25:
The problem is that I cannot roll back the mosquitoes

belenkiy-lab, 2021-05-26:
what prevents you from installing a clean instalation of mosquito not from HA? on any computer on the network or docker?

balbulator, 2021-05-27:
Everything interferes, on the PC not an option, on the docker, too, as on the Home Assistant

belenkiy-lab, 2021-06-03:
[https://1drv.ms/u/s!AgyPTdNGxaf8iY1lVEbLF46puzGa2A?e=vV0QSs](url)

this is my mktt snapshot
as I understand you can restore 5.1.1 using it.
will it help?

balbulator, 2021-06-05:
> [https://1drv.ms/u/s!AgyPTdNGxaf8iY1lVEbLF46puzGa2A?e=vV0QSs](url)
> 
> this is my mqtt snapshot
> as I understand you can restore 5.1.1 using it.
> will it help?

Thanks, I have installed the Mosquitto broker add-on
Current version: 5.1.1, but after HA update (core-2021.6.2) the problem reappeared

DaveTiff, 2021-06-05:
I am on core-2021.6.2 and Mosquitto broker  5.1.1 mine is still working ok

Re Dave

danilkorotkov, 2021-06-07:
this is PubSubClient issue, not AM43

Kite-Smurf, 2021-06-07:
> this is PubSubClient issue, not AM43

Thanks for the updated news on this. The question is how to solve it? and how to get AM43 working with the mqtt 6.0 version?

danilkorotkov, 2021-06-07:
> Thanks for the updated news on this. The question is how to solve it? and how to get AM43 working with the mqtt 6.0 version?

you can include arduino-mqtt lib instead of PubSubClient  lib and then check all of 22 mqtt methods in your/example sketch)
ota update is available
PubSubClient has no update since 2020 so it is possible that it is abandoned

buxtronix, 2021-06-08:
I've not got the bad mosquitto version so can't reproduce. Feel free to send a PR and I'll merge a fix that helps.

MrBigleg, 2021-06-20:
> you can include arduino-mqtt lib instead of PubSubClient lib and then check all of 22 mqtt methods in your/example sketch)
> ota update is available
> PubSubClient has no update since 2020 so it is [...]

Do you mind sharing your sketch? I tried and just go tonnes of errors compiling

CaptInsano, 2021-07-03:
Has anyone got a solution for this issue when mosquitto is already at v6.0.1?

I unfortunately have no mosquitto snapshot at earlier version, all other MQTT integrations are still working great but no action from AM43 integration on my system.

Has anyone successfully ported the code to a different MQTT client?

timdonovanuk, 2021-07-05:
Same issue here. It would help if the serial output had any info or debugging lines about MQTT, as I have no idea whats going on. 

I can see it finding position and battery but nothing is sent to MQTT other than am43/enabled = ON

balbulator, 2021-07-06:
> Same issue here. It would help if the serial output had any info or debugging lines about MQTT, as I have no idea whats going on.
> 
> I can see it finding position and battery but nothing is sent to MQTT other than am43/enabled = ON

exactly the same with me

balbulator, 2021-07-14:
https://github.com/buxtronix/am43/issues/21#issuecomment-880215140
Everything can be cured with ESPHome

Kite-Smurf, 2021-07-15:
> [#21 (comment)](https://github.com/buxtronix/am43/issues/21#issuecomment-880215140)
> Everything can be cured with ESPHome

Could you please share some instructions on how to set this up in ESPHome? I never used that before. Thanks for help.

balbulator, 2021-07-15:
> > [#21 (comment)](https://github.com/buxtronix/am43/issues/21#issuecomment-880215140)
> > Everything can be cured with ESPHome
> 
> Could you please share some instructions on how to set this up in ESPHome? I never used that before. Thanks for help.

Also important in -> Supervisor -> ESPHome -> Configuration ->
esphome_version - buxtronix:am43
certfile - fullchain.pem
keyfile - privkey.pem

```
esphome:
  name: cover
  platform: ESP32
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "any password"

wifi:
  ssid: "Sergio"
  password: "Password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Cover Fallback Hotspot"
    password: "StW7wyE8nD6N"

captive_portal:

esp32_ble_tracker:

ble_client:
  - mac_address: 01:02:03:AA:BB:CC
    id: blinds

cover:
  - platform: am43
    name: "Blinds in bedroom"
    id: cover_bedroom
    ble_client_id: blinds

sensor:
  - platform: am43
    ble_client_id: blinds
    battery_level:
      name: "Blinds battery"
    illuminance:
      name: "Blinds light"
    update_interval: 120s
  - platform: uptime
    name: Uptime Sensor

switch:
  - platform: restart
    name: "Device Restart"
  - platform: ble_client
    name: "Blinds  On"
    ble_client_id: cover
    id: right_on

```

don86nl, 2021-08-06:
> Also important in -> Supervisor -> ESPHome -> Configuration ->
> esphome_version - buxtronix:am43
> certfile - fullchain.pem
> keyfile - privkey.pem
>
> ```
> esphome:
>   name: cover
>   platform: ESP32
>   board: [...]

This helped me, thanks!

sambeetm, 2022-02-02:
Hi..first time with ESPHome. Followed all the above instructions and installed the add-on. Set the following configurations.

esphome_version: buxtronix:am43
certfile: fullchain.pem
keyfile: privkey.pem

My .yaml file is as below and validates. But, throwing bunch of errors during install. Can somebody please suggest whats wrong here?

esphome:
  name: cover
  platform: ESP32
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "anypassword"

wifi:
  ssid: "xxxxxxx"
  password: "yyyyyyy"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ssssssss"
    password: "zzzzzzzz"

external_components:
  - source: github://buxtronix/esphome-am43

captive_portal:

esp32_ble_tracker:

ble_client:
  - mac_address: 02:05:fc:30:41:e3
    id: guest_am43

cover:
  - platform: am43
    name: "Guest Blinds"
    id: guest_cover
    ble_client_id: guest_am43

sensor:
  - platform: am43
    ble_client_id: guest_am43
    battery_level:
      name: "Guest Blinds battery"
    illuminance:
      name: "Guest Blinds light"
    update_interval: 120s

switch:
  - platform: ble_client
    name: "Guest window"
    ble_client_id: guest_am43

Compilation Errors.....

INFO Reading configuration /config/esphome/blind-engine1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing cover (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/cover/.pioenvs/cover/src/esphome/components/am43/am43.cpp.o
Compiling /data/cover/.pioenvs/cover/src/esphome/components/am43/cover/am43_cover.cpp.o
src/esphome/components/am43/cover/am43_cover.cpp: In member function 'virtual void esphome::am43::Am43Component::loop()':
src/esphome/components/am43/cover/am43_cover.cpp:26:27: error: 'Established' is not a member of 'esphome::esp32_ble_tracker::ClientState'
   if (this->node_state == espbt::ClientState::Established && !this->logged_in_) {
                           ^
src/esphome/components/am43/cover/am43_cover.cpp: In member function 'virtual void esphome::am43::Am43Component::control(const esphome::cover::CoverCall&)':
src/esphome/components/am43/cover/am43_cover.cpp:48:27: error: 'Established' is not a member of 'esphome::esp32_ble_tracker::ClientState'
   if (this->node_state != espbt::ClientState::Established) {
                           ^
src/esphome/components/am43/cover/am43_cover.cpp: In member function 'virtual void esphome::am43::Am43Component::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/am43/cover/am43_cover.cpp:97:26: error: 'Established' is not a member of 'esphome::esp32_ble_tracker::ClientState'
       this->node_state = espbt::ClientState::Established;
                          ^
src/esphome/components/am43/am43.cpp: In member function 'virtual void esphome::am43::Am43::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/am43/am43.cpp:33:26: error: 'Idle' is not a member of 'esphome::esp32_ble_tracker::ClientState'
       this->node_state = espbt::ClientState::Idle;
                          ^
src/esphome/components/am43/am43.cpp:56:26: error: 'Established' is not a member of 'esphome::esp32_ble_tracker::ClientState'
       this->node_state = espbt::ClientState::Established;
                          ^
src/esphome/components/am43/am43.cpp:66:18: error: 'millis' was not declared in this scope
           millis() - this->last_battery_update_ > 10000) {
                  ^
src/esphome/components/am43/am43.cpp: In member function 'virtual void esphome::am43::Am43::update()':
src/esphome/components/am43/am43.cpp:95:27: error: 'Established' is not a member of 'esphome::esp32_ble_tracker::ClientState'
   if (this->node_state != espbt::ClientState::Established) {
                           ^
*** [/data/cover/.pioenvs/cover/src/esphome/components/am43/cover/am43_cover.cpp.o] Error 1
*** [/data/cover/.pioenvs/cover/src/esphome/components/am43/am43.cpp.o] Error 1
========================= [FAILED] Took 14.69 seconds =========================

Kite-Smurf, 2022-02-02:
Do not use the external_components:

source: github://buxtronix/esphome-am43

the AM43 is already a built in option in ESP home

panoriega, 2022-04-07:
I put a temp fix for current version using PubSubClient, this is not the best solution, I made this changes just for test purposes.
Now it is able to subscribe and publish topics to mosquitto

I also tried to replace PubSubClient by MQTT 2.5.
Both versions are on box folder and in "working condition", if you want to try and share your comments about it.

https://app.box.com/s/q9us5zsouo36fpt9eri1aietgiv6fwds

Home Assistant 2022.4.0
Mosquitto broker addon 6.0.1

Thanks,
Antonio

danilkorotkov, 2023-04-19:
6.2.0 works perfect